BruceDevices/firmware #522: T-embed hang/slow when selecting Read 125kHz or NRF24

hahmann, 2024-12-06:
I've been using Bruce for just a couple weeks on a Lilygo T-Embed CC1101 and I don't have a 125kHz reader but if I try to use the 125kHz functionality selecting it from the menu, the T-embed starts to work very slowly, the SD card won't show up anymore and I'll have to turn-off / turn-on again to fix this behavior. My expectation would be that if the module is not connect, a message will show up stating it instead of needing to reboot, but not sure if this is possible. 

The same behavior will happen if I select any of the actions under the NRF24 menu without that module installed, T-Embed will hang and start abnormal behavior, super slow on moving from on menu to the other, the battery indicator will move from any number to 100% and will stay there regardless of the % battery left.

Thank you for the great software you guys developed and looking forward to help the community!

Teapot321, 2024-12-06:
There is no 125khz reading in T-Embed

hahmann, 2024-12-07:
I know, another reason to give a message that either not possible or not installed instead of having to restart to fix hanging/slowness. I bought a Chameleon to fix the lack of 125kHz reading.

bmorcelli, 2024-12-07:
fixwd on Beta https://github.com/pr3y/Bruce/commit/3db6bd7ecdc04de792e52a7d6f001d9963533c54

closed 2024-12-07 by bmorcelli