BruceDevices/firmware #523: T-Embed CC1101 - Yellow UI / displayRedStripe with white fonts

hahmann, 2024-12-06:
I have a T-Embed CC1101 and realized that the best UI color for using it outdoors is Yellow and have using it like that successfully.

Unfortunately when the displayRedStripe function is used like reading NFCs, probing hosts, etc the stripe is done using yellow background and white fonts which doesn't provide enough contrast between the two in bright daylight. 

Maybe isn't just with me and we potentially for the Yellow UI use black fonts on the displayRedStripe which will give a better contrast with yellow.

Great project!

Teapot321, 2024-12-06:
As for me, yellow is the worst, white is the best

hahmann, 2024-12-07:
Cool, indeed, tried it know and it is nice that the white uses black fonts for the displayRedStripe, which I think will be better for the yellow as well. Thank you.

rouing, 2024-12-15:
After the UI Colors change, this has been "resolved"
Thanks @bmorcelli, looks better.

bmorcelli, 2024-12-15:
Thank you, you're welcome 🤗

closed 2024-12-15 by bmorcelli