BruceDevices/firmware #529: Esp32-s3 devkit keeps rebooting with bruce firmware

Z-SHOCK-E, 2024-12-07:
error log 

ELF file SHA256: ce15b6de459cf4d7

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40385961
  #0  0x40385961 in spinlock_release at /home/runner/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:155
      (inlined by) vPortExitCritical at /home/runner/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:323

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   559][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[   566][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (18) gpio: gpio_set_level(227): GPIO output gpio_num error
[   572][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (30) gpio: gpio_set_level(227): GPIO output gpio_num error
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420d9262  PS      : 0x00060a30  A0      : 0x820d933c  A1      : 0x3fcebb90  
A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x00000000  
A6      : 0x000000ff  A7      : 0x00000008  A8      : 0x08000000  A9      : 0x3fcebb60  
A10     : 0x3fca8798  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00040000  
A14     : 0x00000000  A15     : 0x3fc9d5e8  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x420e8144  LEND    : 0x420e81a8  LCOUNT  : 0x00000000

bmorcelli, 2024-12-07:
Add this flag to the frey

-DUSE_HSPI_PORT=1

Displays need this flag on Esp32s3 boards

closed 2024-12-07 by bmorcelli

Z-SHOCK-E, 2024-12-07:
Can I still use VSPI for the display after using this flag.

khaleeqmalik786786, 2025-05-09:
where need to write this flag