BruceDevices/firmware #531: Bruce ui won't fit inside the 170x320 st7789 1.9" tft display

Z-SHOCK-E, 2024-12-08:
I got bruce working with my 1.9" tft display. I tried swapping the height and width of the display but the ui won't fit perfectly with the display

rouing, 2024-12-15:
You mean you wanted the orientation to be vertical? This is not going to work.

See this: #537

rouing, 2024-12-15:
@bmorcelli NP, To Close.

closed 2024-12-31 by bmorcelli

pepek23588, 2025-07-25:
> I got bruce working with my 1.9" tft display. I tried swapping the height and width of the display but the ui won't fit perfectly with the display

What board did you use esp 32 d?

Z-SHOCK-E, 2025-11-07:
> > I got bruce working with my 1.9" tft display. I tried swapping the height and width of the display but the ui won't fit perfectly with the display
> 
> What board did you use esp 32 d?

Esp32s3 devkit