BruceDevices/firmware #542: The problem with copying SubGhz signals

Teapot321, 2024-12-10:
Since the recent beta, where the menu for copying signals was changed, RAW signals began to be copied more heavily, in the past I pressed the button on the key and the signal was instantly copied, now I press a couple of times while holding the key close and it may not be copied (the spectrum works even at 10 meters). Please fix this and also consider #528.

bmorcelli, 2024-12-11:
Hey man.. I was thinking about that today... I changed the way Raw capture works in order to reduce the noise capturing, but it isn't good enough...

Actually, I need some help to figure out a good solution for it... And the algorithm for that is within this function: https://github.com/bmorcelli/rc-switch/blob/13c3aaf0da8d5fe2a568f1e0a38c69cb23c87ea0/RCSwitch.cpp#L701

The RAW variables..

The other Issue I think it is considerable now... Inwill implement it in my fork until I finish other things

Teapot321, 2024-12-11:
Add spectrum when recording and sending RAW signals, as in flipper zero.

[image]

bmorcelli, 2024-12-11:
This is hard to make, it will take a loooooong time..

rouing, 2024-12-15:
The current "spectrum" display really sucks. :(

bmorcelli, 2025-02-17:
will be followed in here https://github.com/pr3y/Bruce/issues/804

closed 2025-02-17 by bmorcelli