BruceDevices/firmware #544: Support for other chip/modules other than ch9329 for bad usb

jakre1234, 2024-12-11:
Hi devs,
wondering if it is possible to use other UART to HDI modules other than the ch9329 for the bad usb as the ch9329 based module is nowhere available in a lots of region.

Thanks and Regards,

bmorcelli, 2024-12-12:
if they are UART to HID, probably they can work with bruce, once UART communication might be standard to all devices that uses it..

jakre1234, 2024-12-16:
> if they are UART to HID, probably they can work with bruce, once UART communication might be standard to all devices that uses it..

I am trying to find modules which can be used, chatgpt suggests to use :
```
1. CP2110 or CP2112
2. ATmega32U4-based Modules (e.g., Arduino Pro Micro)
3.  RP2040-based Modules (e.g., Raspberry Pi Pico)
4. CH340G or CH341A
5. FT260 (FTDI USB-to-I²C/UART Bridge with HID)
6. Digispark (ATTiny85-based)
```
But I am guessing these will require some external code or changes in the firmware

jakre1234, 2024-12-16:
I have got Digispark ATtiny85 which I am currently using for bad usb by flashing ducky scripts directly to the module, not sure how this will work with m5stick c plus 2 and uart connections or will require any changes in the firmware?

bmorcelli, 2024-12-16:
Digistark would need a side development, to create a firmware that can receive UART on one side and deliver HID in the other.. in the research l did I couldn't find a way to do that quite well due ti hardware restrictions, but I confess I didn't spend too much time on that too

The others I don't know.. I don't have them, and won't purchase myself.. so I encourage you to give it a try..

RP2040, AtMega would need a side development 

Ft260 would need changes in bruce code, too hard to make imo..

The others IDK

jakre1234, 2024-12-16:
Thanks @bmorcelli,
Found the ch9328 available here, will give it a shot...hopefully it will work else I can find the ch9329 chip(no module) and would require a DIY to make uart to usb module(if you can help with that)

https://cdn-learn.adafruit.com/assets/assets/000/130/514/original/CH9328DS1.PDF?1717789673

Thanks and Regards,

closed 2024-12-31 by bmorcelli