BruceDevices/firmware #549: Bug in recording SubGhz signals

Teapot321, 2024-12-13:
When entering Scan/Copy, RAW is immediately written, it is impossible to stop recording RAW until some signal is recorded. That is, there is no Stop RAW button when entering, it appears after recording the signal.

rouing, 2024-12-15:
This is expected behavior. If you want to stop capture and you havent captured something yet, just back out of the application. The only reason you would stop capture in that application is if you have captured your signal and are ready to do something with it.

synl0rd, 2024-12-17:
Why does the full RF jammer feature only last for a short time? It runs for a few seconds and then stops, returning to the main menu. Could a mode be created to allow jamming to continue indefinitely until a button is pressed to stop it?

bmorcelli, 2024-12-17:
Bruce is meant to educational purpose, and SubGhz jamming is a Crime (or can lead into it) in most countries, and we don't want people using our firmware to run into other peoples houses/companies or jamming someone elses car to steal things... So, 20 seconds os enough for studying and understanding of the risks...

This Jammer won't be unleashed in the main project.

varya33, 2024-12-18:
> Bruce is meant to educational purpose, and SubGhz jamming is a Crime (or can lead into it) in most countries, and we don't want people using our firmware to run into other peoples houses/companies [...]

Please tell me where I can reduce the maximum operating time, I want to set it to 5 seconds

bmorcelli, 2024-12-18:
> Please tell me where I can reduce the maximum operating time, I want to set it to 5 seconds

Aham.. sure.. 🫡

varya33, 2024-12-18:
> Aham.. sure.. 🫡

honestly)

bmorcelli, 2024-12-18:
I believe you.. and my name is Jack Sparrow 🏴‍☠️