BruceDevices/firmware #551: iOS Spam

xTyzen, 2024-12-13:
Good Afternoon ! I recently had a bug with iOS Spam in the BLE category. When I spam BLE for iOS devices it appears 1/2. So there are times when it doesn't work, but times when it does. When it spams, it only spams one device. So there is always a device visible on my iPhone. It only appears once every time. And not several times. Thank you for your understanding! I am writing this message to help you improve BLE and above all to help you find out where the problem is. And make it less frustrating for other users. Good day !

dghlkxyvdyxj, 2024-12-15:
apple patched ios popup spamming on ios 18

closed 2024-12-16 by bmorcelli

jakre1234, 2024-12-16:
Not really, the ios spam and android spam works perfectly with flipper zero