BruceDevices/firmware #554: CYD-2432W328C(inv_colors)_and_CYD-2432S024C

H4NT3R0, 2024-12-14:
**Describe the bug:**

STUCK IN LOADING SCREEN

**Steps To Reproduce:**

- Device used: 
- ESP32 CYD-2432S024

**Expected behavior:**

**Actual behavior:**

**Additional context / logs:**

bmorcelli, 2024-12-16:
Can you please send Serial Monitor Logs?

JERGENT, 2024-12-17:
ESP32 CYD-2432S024

> Can you please send Serial Monitor Logs?

closed 2024-12-31 by bmorcelli

Eifeow, 2025-01-18:
I have the same problem I'am stuck at the loading screen even the shark doesn't popped up. What should I do?

bmorcelli, 2025-01-18:
Sending a Serial Monitor log helps a lot

Eifeow, 2025-01-19:
[ 23617][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
This pops up in the 
[ 26352][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop ret i2cWriteReadNonStop returned Error -1
[ 20121][E][Wire.i2cWriteReadNonStop returned Error -1
[ 26366][E][Wire.cpp:499] reques[E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[ 20136][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[ 29659][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[ 29666][E][Wire.cpp:499] requestFrom(): i2cWriteRea-1
[ 35911][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop ret i2cWriteReadNonStop returned Error -1
[ 29681][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[ 29688][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop return[Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
Stop returned Error -1
[ 29702][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[ 29710][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[ requestFrom(): i2cWriteReadNonStop returned Error -1
messeages like these pops up.