BruceDevices/firmware #555: The device doesn't wake up after sleep on the ESP32-2432S028

TheMiner203, 2024-12-14:
**Describe the bug:**
Touching the screen does not wake the device from sleep on ESP32-2432S028

**Steps To Reproduce:**
Wait until the device goes into sleep mode, then touch the screen
- Device used: 

**Expected behavior:**
After touching the screen, the device wakes up

**Actual behavior:**
After touching the screen, the device does nothing

**Additional context / logs:**
¯\_(ツ)_/¯

shnmorimoto, 2024-12-16:
In fact, the entire screen is not touch area. Only the bottom of the screen supports touch. I think you can wake up from sleep by touching the bottom of the screen.

TheMiner203, 2024-12-16:
> In fact, the entire screen is not touch area. Only the bottom of the screen supports touch. I think you can wake up from sleep by touching the bottom of the screen.

Maybe, but it's very inconvenient and not obvious.

closed 2024-12-31 by bmorcelli