BruceDevices/firmware #562: [T-Embed-CC1101] WS2812B RGB LED Support

rouing, 2024-12-16:
Within the encoder, there is 8x WS2812B LED's with the data attached to "IO14"
Right now we do nothing with it so it would be cool to start implementing it as a status indicator or whatever.

This is an Enhancement request that will serve as the starting point of the work.

Questions to answer so far:

What other devices which Bruce supports have Addressable LEDs on them?
How many LEDs each? 
^ Must be answered because the implementation must be generic in that case (In some ways)

Has there been any code written to this effort? 

# Other Notes
platformio.ini in the lilygo-t-embed-cc1101 has a definition "RGB_LED" pointing to pin 21. 
"RGB LED runned by xylopyrographer/LiteLED@^1.2.0 library" Is this in use anywhere?

(Most of these questions are for me to answer and for my personal tracking benefit)

rouing, 2024-12-16:
led.cpp is gonna need some work.

The HAS_RGB_LED needs to be toggled on for it to work but it still wont work as intended since led.cpp is written for a single LED. We want to do way more than that

rouing, 2024-12-16:
Additionally, the only device making use of this appears to be the M5Stack Cardputer.

rouing, 2024-12-16:
Ive had to change to the FastLED Library

bmorcelli, 2024-12-16:
>What other devices which Bruce supports have Addressable LEDs on them?
How many LEDs each?

Only Cardputer has it, with only One RGB Led controlled by one pin

> Has there been any code written to this effort?

I don't think so 

The only device with more and addressable LEDs ir T-Embed (all versions)

rouing, 2024-12-17:
Inital release of functionality is ready. The first feature PR has been marked ready for review. This is not a complete feature yet but it functions as it was when I first started on it with additional support for the T-Embed-CC1101.

mattbad04, 2024-12-17:
How can I flash to my t embed ? With rgb support

EnygmaINC, 2024-12-23:
> How can I flash to my t embed ? With rgb support

Same question

rouing, 2024-12-23:
I have another priority PR waiting before I finish the development on this feature

bmorcelli, 2024-12-31:
can it be closed?

rouing, 2025-01-02:
We can close this one

closed 2025-01-02 by rouing