BruceDevices/firmware #566: ADD 57600 bps to GPS config settings.

ALoGeNo, 2024-12-16:
Hello guys, I did reversed serial port in an old GPS (from 2006, it was getting dust over there) [Royaltek RBT-1000](https://www.manualslib.com/manual/142744/Royaltek-Rbt-1000.html?page=9#manual) here is the [schematics ](https://fcc.report/FCC-ID/RCCBT/428871.pdf).

It works at 57600 bps, so I guess that maybe more devices use this baudrate, maybe you can add this value in the settings for get more supported devices.

I did modify the code my self and it is working now over GROVE port.

[gps57k_0](https://github.com/user-attachments/assets/13566b8f-2d77-4b90-be2c-78e0e97b6e9c)

[image]

Thanks in advance.

rouing, 2024-12-16:
I added 2 known GPS Baudrates. 57600 being one.

bmorcelli, 2024-12-17:
Merged, on Beta, will be available in the next release

closed 2024-12-17 by bmorcelli