BruceDevices/firmware #569: [Global] USB OTG Mounting of Internal FS and SD Card

rouing, 2024-12-17:
It would be nice to have the ability to mount the SD Card (and potentially, the Internal FS) over USB so that things can be added to it easily. 

### Questions to be answered:

- Has any work been done to this effort already?
- Are there examples we can use?
- Which devices would support this? Which devices wont?
- Do we automatically mount the FS on connection or only if its toggled on/off in the Config? 
- Where in the UI would we put such settings?

- File Layout and Structure? Location?
- Will we need any additional libraries for this?
- Is there a library we can use to make this easy?

rouing, 2024-12-17:
[USB MSC - Espressif Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/usb_msc.html)

bmorcelli, 2024-12-17:
I have searched for that in April to use it on Launcher, and unfortunetly couldn't find anything..

what I know is:
* ESP32 doesnt have OTG interface, and their USB plugs are connected to a dedicated UART->Serial converter that can't be used for other kind of purpose.. 
* ESP32-S3 usually have OTG and the USB can be used as other purposes such as HID etc, but I don't know if it can in fact read and write from flash drivers with good performance (https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/msc), (https://esp32.com/viewtopic.php?t=35266), and in this case there are no Arduino Library for that purpose.

rouing, 2024-12-19:
There is a way to do this as ive seen it done before. Its surprisingly hard to find a quick example that allows for exposing the SD Card, not just the internal flash.

Also: Yeah this can only be done on the ESP32-S3. I do hear the performance is basically USB1.1 but if were just uploading small files, that shouldnt be an issue.

IncursioHack, 2025-03-21:
Done!
https://github.com/pr3y/Bruce/pull/891

closed 2025-03-21 by IncursioHack