BruceDevices/firmware #570: [Global] (RF) Implementation of support for parsed protocol .sub files

rouing, 2024-12-17:
Right now, we can only use RAW (And I think a few others.. maybe? Have to check) protocol in .sub files. This request comes in 2 parts, one is covered in another issue #192.

The first part is processing of a .sub file to transmit that contains a parsed protocol option (Megacode as an example)
The second part is capturing, identifying, parsing, and processing of a signal during capture/receive and then saving that signal in its parsed format as a .sub file.

This issue is a stub for now.

rouing, 2024-12-17:
@eadmaster Are you still around and are you still working on your port?

rouing, 2024-12-17:
There is also tons of room for improvement on signal capture all together. At the moment I have to be really close to the device to capture a signal and it is usually pretty mangled and unusable.

eadmaster, 2024-12-17:
> This request comes in 2 parts, one is covered in another issue https://github.com/pr3y/Bruce/issues/192.

Sensors decoding with rtl_433_ESP is a completely different feature, not related to generic RF signals dumping/replaying.

> At the moment I have to be really close to the device to capture a signal and it is usually pretty mangled and unusable.

If the signal gets decoded by the RCSwitch lib, it should be more reliable than a RAW capture.
To get a better range you may need to replace the antenna.

rouing, 2024-12-17:
> > This request comes in 2 parts, one is covered in another issue #192.
> 
> Sensors decoding with rtl_433_ESP is a completely different feature, not related to generic RF signals dumping/replaying.

I understand that and I sort of slipped something in there I am half hoping to achieve here at a later point. It would require generation of a sub file from a whole new process itself. 

> > At the moment I have to be really close to the device to capture a signal and it is usually pretty mangled and unusable.
> If the signal gets decoded by the RCSwitch lib, it should be more reliable than a RAW capture.
> To get a better range you may need to replace the antenna, or reduce environment noise.

Alright. Ill adjust my testing accordingly. I did slap on a new antenna myself but have found the SMA connector has some loss. Ill play with that later. 

Heres some funny results of my testing so far anyways.
What it was vs what it shouldve been:
[image]
[image]

bmorcelli, 2024-12-17:
These graphics are from what? HackRF? Rtl-SDR?

I want to buy one for me as a Christmas gift, need advices