BruceDevices/firmware #574: Problemas com BadUSB

igoredusva, 2024-12-17:
**Describe the bug:**
o comandos "SPACE" do BADUSB estão sendo substituidos por "," quando executado (mesmo com layout Portugues Brasil);
"SPACE" -> ","
        "ú" -> "]"
        "ç" -> "2"
        "ã" -> "x"
         "|" -> ""

**Steps To Reproduce:**
Others>BadUSB
rodar script de teste com um bloco de notas aberto para visualização, ou qualquer outro script que possua o comando "SPACE" ou caracteres especiais.

- Device used: M5 Cardputer - Bruce 1.8

**Expected behavior:**
Com o layout correto selecionado (Portugues Brasil)
Esperavasse que os caracteres acentuados funcionassem perfeitamente, além do comando "SPACE" funcionar para inserir um espaço

**Actual behavior:**
Informado da descrição do bug

**Additional context / logs:**
Script para teste (delay inicial de 5s para deixar aberto o bloco de notas para visualização do teste)
[teste.txt](https://github.com/user-attachments/files/18170761/teste.txt)

bmorcelli, 2024-12-17:
Qual é o Layout do teclado do teu computador?

igoredusva, 2024-12-17:
ABNT2

closed 2025-05-26 by bmorcelli