BruceDevices/firmware #576: CC1101 does not work with Bruce

chelick5, 2024-12-18:
I encountered a problem that on M5stickc plus2 with firmware Bruce 1.8. if you connect the cc1101 module to the device, and select RF CC1101 in the configuration. then the device freezes on the module selection menu and does not respond to buttons. If in this state you disconnect any wire, the screen will display the message “CC1101 not found”. with old firmware versions everything is the same. If you can help I will be very grateful

varya33, 2024-12-18:
I have a similar problem when starting a full jamming it crashes into the main menu

chelick5, 2024-12-18:
Many people have problems with cc1101

varya33, 2024-12-18:
The developer changed something in the code and now jamming full only works for 10 seconds

bmorcelli, 2024-12-18:
> Many people have problems with cc1101

Many people don't read the wiki
Many people can't follow video instructions 
Many people buy crappy CC1101 boards or burn them connecting it wrong
Many people don't want to learn the basics

So yes, many people might have problems with it.

As I have it up and running, the problem is not on the firmware, it is an 8th OSI layer model problem..

closed 2024-12-18 by bmorcelli