BruceDevices/firmware #578: Cc1101

chelick5, 2024-12-18:
how to solve the problem that the cc1101 module does not work when connected and the module selection menu freezes? I apologize for bothering you a second time, I just don’t understand what needs to be done to get the module to work.

bmorcelli, 2024-12-18:
https://youtu.be/RVQrfghGVqw?si=4nw9nrrikceI3JiC

Push the wires into the grove connectors (push the wires into the dupont plastic)

You can take the dupont connector out, press the metal parts to tighten up the hole, making the connection better.. you can isolate the metal parts and use like this, or put the plastic back on..

The grove pins are thicker than Male Dupont connectors, so the problem you are facing is due to false contact...

[image]

[image]

[image]

Good luck.

closed 2024-12-18 by bmorcelli

M5stick-c-plus2-guy, 2025-06-13:
Wich Part Is the terminal?