BruceDevices/firmware #579: M5stickc plus2 cc1101

chelick5, 2024-12-19:
Hello, it's me again. with a question about the cc1101 module. You recommended that I remove the plastic from the wire and crimp it. and surprisingly it helped, the module was detected. but a new problem has appeared. when I press either the jammer or scan/copy everything freezes again, although the wires have good contact. I would be grateful if you give an answer.
I apologize for writing this 3 times already.

bmorcelli, 2024-12-19:
It can  be faulty connections, or faulty module, or misuse.. 

Send a video or a Serial Monitor Log, otherwise there are no way to help..

And next time write on the same issue, don't need to open a new one, look for the closed ones..

chelick5, 2024-12-20:
https://imgur.com/gallery/4gHvN4c