BruceDevices/firmware #585: [T-Embed-CC1101] Battery indicator does not rise above 50%

shnmorimoto, 2024-12-21:
**Describe the bug:**

Even after charging, the battery indicator does not rise above 50%

**Steps To Reproduce:**

- Device used: T-Embed-CC1101
- Beta Release（There was no issue with 1.8.）

**Expected behavior:**

Charged to 100%

**Actual behavior:**

Charging seems to stop at around 50%
It may just be a display issue

**Additional context / logs:**

EnygmaINC, 2024-12-23:
Same. Above ~40% for me.

shnmorimoto, 2024-12-23:
Even with factory firmware, charging stops halfway through.

I'm not sure, but the full capacity is set to 3000. Is that correct?

It should be 1300mAh?

[image]

shnmorimoto, 2024-12-25:
It displayed normally after I flashed the firmware several times.
I guess it was a problem with my device.

closed 2024-12-25 by shnmorimoto