BruceDevices/firmware #586: CYD (2432S028) Support for wiring multiple peripherals - CC1101, NRF24, PN532, and maybe an SPI sd adapter to work simultaneously

particleacceleratorbudapest, 2024-12-21:
Hi there, this is my first 'issue' so forgive me if i do something stupid 

So i would love to build a CYD that can do it all, like all the menu listed features. 

I will make separate issues to solve here : 

1. micro SDCARD slot

I bought an SPI MicroSD card adapter (HW-125) in order to minimize the width requirements for a case, and maybe use something new instead of the tipical red microsd sniffer, as its looks pretty dumb to build the case around it. Got an sd to microsd adapter cable, but that still hangs out on the side, so yes i would appreciate some sort of solution instead of the sniffer.

First question, can an SPI microsd adapter work with all the above mentioned peripherals instead of microsd sniffer ? 
IF yes, how to wire it up so it wont have issues with other peripherals ? It has the standard CS, SCK, MOSI, MISO, VCC, GND pins . 

2. NRF 24 and CC1101 

I would like to know whether is it already sorted out to have multiple peripherals running on the CYD like nrf24, cc1101 and pn532 ? 
IF yes, i would really appreciate some sort of wiring instructions when ALL of these peripherals connected at the same time. Can i connect multiple wires to the same ( 22, 27 pin ) and to the sd adapter, will it work like that ? 

3. Does all the above mentioned peripherals interfere with any sd card related processes like save files, load files from the microsd ? since like all of these peripherals have to be connected to the microSDcard slot 

Thank you VERY much for all the efforts You guys already made possible with this config, i love it, just installed 1.8 bruce, love the animated splash screen ! :) Thumbs up and have a great day y'all! Respect from Hungary

Neo2a, 2025-03-06:
Kaptál már választ a kérdésedre?

samexit23, 2025-04-02:
i dont really know a lot but for the nrf24 and cc1101 you can just put switches to their vcc and there will be no problem

closed 2025-05-24 by bmorcelli