BruceDevices/firmware #595: Problem saving a signal

MAR1TO15, 2024-12-22:
When I capture a signal, from a ringer in this case and reproduce the newly captured signal it works, but when I save it and try to play it from the .sub file it doesn't work
I'm using an M5StickC plus 2 and a C1101 with Bruce 1.8

samuelstj, 2024-12-22:
Hi. I just faced the same issue.

I also tryied to replay an old .sub and it isn't working.

Some times it saves in the RcSwitch protocol but the key is empyt.

I'm using cardputer with M5 RF modules and Bruce 1.8

JerikoJay, 2024-12-23:
Yes, I have the same problem on Bruce 1.8 firmware (m5 stick P 2)

eadmaster, 2024-12-29:
this could be related to rolling codes , see the last point if this faq
https://github.com/pr3y/Bruce/wiki/FAQ#irrf-transmitterreceiver-not-working

bmorcelli, 2024-12-31:
Can you please test the last Beta Release?

bmorcelli, 2025-01-20:
Fixed in 1.8.1

closed 2025-01-20 by bmorcelli