BruceDevices/firmware #597: Custom IR not working with T-Embed CC1101

orion2036, 2024-12-22:
Hi.
I manage to copy the on/off signal of my Smart Tech tv with "IR Read" but i can't use "Custom IR" to shutdown my TV. It is sending an IR signal but nothing happen, i can't turn off or turn on my tv. (TV-B-Gone can't turn off my tv too but this is another issue, maybe my device is not present on the IR database).
Maybe a trouble with the NEC42 protocol?

Details about the file used : 
type : parsed
protocol : NEC42
address : 00 7F 00 00 
command : 15 00 00 00 
bits : 32
value : 00 FE A8 57

Thank you.

j0124a, 2024-12-23:
Yes this seems to be an issue.
 Tv-B-Gone works  for me as well as some raw signals.  Anything from the .ir code library (such as https://github.com/Lucaslhm/Flipper-IRDB ) or "IR read" does not work.

Lamnxzp, 2024-12-23:
@j0124a
> Anything from the .ir code library
Does it apply to .ir codes with protocols other than NEC?

j0124a, 2024-12-23:
Sony does not work either.

Lamnxzp, 2024-12-23:
> Sony does not work either.

You mean `SIRC`, `SIRC15` or `SIRC20`? It should be written after "protocol: " in the IR file

orion2036, 2024-12-23:
> @j0124a
> 
> > Anything from the .ir code library
> 
> Does it apply to .ir codes with protocols other than NEC?

I don't have issue with others tv. Somebody in Discord have trouble with NECext protocol too.

j0124a, 2024-12-24:
> > Sony does not work either.
> 
> You mean `SIRC`, `SIRC15` or `SIRC20`? It should be written after "protocol: " in the IR file

SIRC - for example :

name: Power
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 15 00 00 00

Lamnxzp, 2024-12-24:
https://github.com/pr3y/Bruce/blob/bfcc813dc2b07d5388c1635272371454a4e7c691/src/modules/ir/custom_ir.cpp#L406-L420
`irsend.sendSony` uses `20 bits` (hardcoded in the code), so I think it is the problem.
Maybe by editing (at line 417), `20` to `nbits` it would work 🤷

By the way, another issue here, if `nbits` is `20`, `encodeSony` needs the `extended` part, which is not provided here.

Lamnxzp, 2024-12-24:
@j0124a You are using `SIRC` so it is `12` bits, and here it is sending `20` bits.
I think it is the issue.

Lamnxzp, 2024-12-24:
Another thing, I wonder why there are  `10` repeats, I'm pretty sure SIRC needs only `2` additional repeats (`1 initial + 2 repeat`).

Lamnxzp, 2024-12-24:
I made the changes, and it seems to work on my side. I get the right values 👍

bmorcelli, 2024-12-24:
Great! When you have the changes ready and tested, open a Pull Request to submit these fixes

Lamnxzp, 2024-12-24:
I tested it on my Sony TV, and it works 👍
It works for 12 bits SIRC, not sure for 15 bits and 20 bits.

j0124a, 2024-12-24:
If i read IR directly from the remote it gives this ( with bits:12 ).   Replay of any ir code NEC etc also does not work even if the IR code is correct . 
Filetype: Bruce IR File
Version: 1
#
# MyDevice
name: Btn0
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 15 00 00 00
bits: 12
value: 00 00 0A 90
#

Lamnxzp, 2024-12-24:
@j0124a I found the issue and it has been fixed here #604

Lamnxzp, 2024-12-24:
@j0124a Can you send here a .ir file with protocol NEC that you got using IR Read with Bruce, that isn't working when you send the command?

bmorcelli, 2024-12-24:
The fix for it is available on Beta for testing..

@Lamnxzp thank you for your fixes

Lamnxzp, 2024-12-25:
> @Lamnxzp 
> 
> 
> Filetype: Bruce IR File
> Version: 1
>  MyDevice
> name: Btn0
> type: parsed
> protocol: NEC42
> address: 00 7F 00 00
> command: 15 00 00 00
> bits: 32
> value: 00 FE A8 57
> 

This isn't NEC42, I think there is an issue there. Anyway it should be fixed on beta. Can you try @orion2036 ?

bmorcelli, 2024-12-25:
It is waiting for protocol be ==NEC
https://github.com/pr3y/Bruce/blob/fdf59ffb15ea504432412e1e608ed8f69119739b/src/modules/ir/custom_ir.cpp#L251

I'll change it here to accept this one.. just a sec

Lamnxzp, 2024-12-25:
> I added it because it doesn't work for me in the new beta

The command you sent, is the one you captured from your remote, and when you send it, it doesn't works?

bmorcelli, 2024-12-25:
I just changed ot and SIRC, hope it works...

Available on Beta

bmorcelli, 2024-12-25:
> @bmorcelli I sent PR fix ir read

@hunt3rbb666NotFound  Merged now, itll be available in about 10 minutes

orion2036, 2024-12-28:
> > @Lamnxzp
> > Filetype: Bruce IR File
> > Version: 1
> > MyDevice
> > name: Btn0
> > type: parsed
> > protocol: NEC42
> > address: 00 7F 00 00
> > command: 15 00 00 00
> > bits: 32
> > value: 00 FE A8 57
> 
> This isn't NEC42, I think there is an issue there. Anyway it should be fixed on beta. Can you try @orion2036 ?

Hi.
I am sorry i don't know how to do it, i have limited skills with coding.

PR #635 (2024-12-31): NECext Support, Fix Protocols
#### Proposed Changes ####
- Added support for `NECext` protocol.
- Fix `Samsung32`, `RC6` protocol.
- Fix incorrect `repeats` number (This resolves the issue where the command was being sent multiple times unnecessarily).
- Removed `NEC42` from the README, because it is not supported.

Fix #597
closed 2024-12-31 by bmorcelli