BruceDevices/firmware #605: FM functions are not working with CJMCU-4713

sanchez8096, 2024-12-24:
**Describe the bug:**
After correctly connecting the module and choosing any FM functions, firmware outputs "Cannot find radio".

**Steps To Reproduce:**
1. Connect CJMCU-4713 to grove pins on Cardputer as described in this repo.
2. Short RST and GND to enable module.
3. Go into FM functions and select any of them.
4. Error "Cannot find radio".

**Device used:**
M5 Cardputer
FM module CJMCU-4713: https://sl.aliexpress.ru/p?key=Xi0sGk9

**Expected behavior:**
FM functions working

**Actual behavior:**
FM functions are not working

**Additional context / logs:**
Connections are soldered and tested.

The thing in question:
[image]

bmorcelli, 2024-12-24:
Try changing the switch beside the grove to 5VOUT position (towards the grove)

sanchez8096, 2024-12-24:
> Try changing the switch beside the grove to 5VOUT position (towards the grove)

It was already in that position. Also tried in 5VIN position, didn't work.

sanchez8096, 2024-12-26:
Update: it works on M5 Stick Plus2. Connections stayed the same, I just plugged grove cable in Stick.

closed 2024-12-30 by bmorcelli

Grzybekrk, 2025-01-17:
Is this works on cardputer ??

romacoma11, 2025-03-25:
> Update: it works on M5 Stick Plus2. Connections stayed the same, I just plugged grove cable in Stick.
how could you connect this to a stick

romacoma11, 2025-03-25:
how could you connect this to the stick

romacoma11, 2025-03-25:
> Update: it works on M5 Stick Plus2. Connections stayed the same, I just plugged grove cable in Stick.

how did you connect it to a stick