BruceDevices/firmware #606: Backlight on button area

Viarety, 2024-12-25:
Hello, as far as I know, the t-embed has a built-in backlight in the button area. By default, there is a function in the firmware to turn on and adjust this backlight, do you plan to do this in the Bruce firmware as well? I think many people would like to see this feature in the new update.

bmorcelli, 2024-12-25:
It is being developed here https://github.com/pr3y/Bruce/pull/563

Viarety, 2024-12-25:
There's a lot of information there, and it's not very clear how to turn on the backlight.

Viarety, 2024-12-25:
Can you provide me with something like a guide on how to do this?

bmorcelli, 2024-12-25:
you wait ;)

Viarety, 2024-12-26:
pls

rouing, 2024-12-28:
> pls

Its currently being developed. The feature has not been released yet. Check the linked PR to see why:

              Pending #571 merge to fix up the flags

_Originally posted by @rouing in https://github.com/pr3y/Bruce/issues/563#issuecomment-2556722250_

bmorcelli, 2024-12-31:
Available now on Beta

closed 2024-12-31 by bmorcelli