BruceDevices/firmware #608: Rf replay not working m5stick cplus 2

thanatos84, 2024-12-25:
**Describe the bug:**
RF replay not working on m5stick C plus 2

**Steps To Reproduce:**
Select all ranges, intercept signal and reply. Neither works if I save it and then I try to replay it
- Device used:  m5stick c plus 2

bmorcelli, 2024-12-25:
It was fixed in this commit https://github.com/pr3y/Bruce/commit/e92a3abd8b116c7cc754bbc7405c32e77e0fb4ef

please test it with the last beta

thanatos84, 2024-12-25:
Tried with latest beta in the website, but still not working. I've tried with my garage door and also my car, but not working. But it works fine with my hackRf, so it's not due to a problem with a rolling code

I'm using CC1101. Spectrum and jammer works fine, but when I go to Scan/Copy, it reads the signal, it saves the file. But if I click on replay it doesn't work, neither if I go to "main menu > rf > custom subhhz > select file", it appears "sending" but nothing happens

bmorcelli, 2024-12-26:
Ok, I'll keep looking into the Raw signal detection/saving.. it'll take some time tho

bmorcelli, 2024-12-26:
As you have a HackRF, can you check one thing for me?

I believe the signal is being saved "inverted"... Like high times and Low times swapped...

If this is the problem, I might need to add one dummy time at the beginning to invert it..

If you can send a snapshot of both original signal and replayed signal, where we can see the first 10 square waves, maybe it can help me debugging this issue

MAR1TO15, 2024-12-26:
I don't know if it has anything to do with it, but checking the .sub file, it is missing ":" after the version, so no .sub file processor accepts it.

bmorcelli, 2024-12-26:
It is not the problem.. this version section is nothing but a comment

bmorcelli, 2024-12-29:
Today I could successfully open my gate with the RAW data collected with Cardputer and T-Embed CC1101 with this last version available on Beta

@rustysun9 if you could do the testing please

thanatos84, 2024-12-29:
> Today I could successfully open my gate with the RAW data collected with Cardputer and T-Embed CC1101 with this last version available on Beta
> 
> @rustysun9 if you could do the testing please

Wich version? Because this afternoon I updated last beta and I just received dozens of fake raw signals in m5stick cplus2

I thought may be there was some noise in the air, but I checked with flipper zero also and nothing. Also I installed latest stable and no noise (but not working for sure)

thanatos84, 2024-12-29:
I've installed latest beta and device is not booting, I've had to install again latest stable

rouing, 2024-12-29:
> I just tested with the latest beta my 433mhz car remote, it captures the raw signal but the replay doesn't work. I also tried saving more signals but no luck :(

You mean your Cars Keyfob? Please dont try this. You can cause issues and lock yourself out of the vehicle. Its also not likely to work.

MayLaOwn, 2024-12-29:
@rouing Flipper Zero works the replay without causing damage. Isn't this possible with Bruce? Meaning I can't carry out RF tests on my car?

MayLaOwn, 2024-12-29:
@rouing Flipper Zero works the replay without causing damage. Isn't this possible with bruce? Meaning I can't carry out RF tests on my car?

rouing, 2024-12-29:
> @rouing Flipper Zero works the replay without causing damage. Isn't this possible with bruce? Meaning I can't carry out RF tests on my car?

You can cause your keyfob to get out of sync with the car. Then you will be locked out.

MayLaOwn, 2024-12-29:
@rouing I didn't understand the answer and for that I apologize. can you explain me better? Raw signal replay should work and cause no damage...

rouing, 2024-12-29:
https://tcm-sec.com/intro-to-car-hacking-replay-attacks/

"Some of these attacks may put a key fob and car out of sync, causing the key fob to no longer work until the car and fob are reprogrammed."

Everything you need to know is in that article.

rouing, 2024-12-29:
Bruce should still be working yes though, were having issues apparently.

thanatos84, 2024-12-29:
What devices are you using? I've tried again to flash via the website latest beta in m5stic cplus 2 and not booting, it just turn on green led when press power on… but not booting

Then I go back to stable 1.8 and boots perfectly

MayLaOwn, 2024-12-29:
@rouing Ok now at least we know not to even try. Hopefully this will be different for Bruce in the future, even if it means risking my devices I should be able to do it. Thanks for the reply!

eadmaster, 2024-12-30:
It may worth having separate menu entries for raw and decoded signals replay. 

I've found sometimes the RcSwitch library skips a part of the signal for some reason.

For instance, this is the output of `rtl_433 -a 4` with a smart home plug i have:

````
[00] {151} 4c 0f 08 26 07 84 13 03 c2 09 81 e1 04 c0 f0 82 60 78 40 
[01] {33} a1 82 98 93 00                                           : 10100001 10000010 10011000 10010011 0
[02] {33} a1 82 98 93 00                                           : 10100001 10000010 10011000 10010011 0
[03] {33} a1 82 98 93 00                                           : 10100001 10000010 10011000 10010011 0
[04] {32} a1 82 98 93                                              : 10100001 10000010 10011000 10010011 
````

the initial row (`[00] {151} 4c 0f...`) is not present in the generated sub file.

bmorcelli, 2024-12-30:
> It may be worth having separate menu entries for raw and decoded signals replay.
> 
> I've found sometimes the RcSwitch library skips a part of the signal for some reason.
> 
> For instance, this is the output of `rtl_433 -a 4` with a smart home plug i have:
> 
> ```
> [00] {151} 4c 0f 08 26 07 84 13 03 c2 09 81 e1 04 c0 f0 82 60 78 40 
> [01] {33} a1 82 98 93 00                                           : 10100001 10000010 10011000 10010011 0
> [02] {33} a1 82 98 93 00                                           : 10100001 10000010 10011000 10010011 0
> [03] {33} a1 82 98 93 00                                           : 10100001 10000010 10011000 10010011 0
> [04] {32} a1 82 98 93                                              : 10100001 10000010 10011000 10010011 
> ```
> 
> the initial row (`[00] {151} 4c 0f...`) is not present in the generated sub file:
> 
> ```
> Filetype: Bruce SubGhz File
> Version 1
> Frequency: 433919968
> Preset: 2
> Protocol: RcSwitch
> Bit: 32
> Key: 00 00 00 00 A1 82 98 93
> TE: 706
> ```

This is the preamble part that usually is prepared by the `Preset: 2` in the RCSwitch library... 

The raw reading is reading the timings completely... You can activate RAW Only if you have DevMode activated and try to catch and replay it..

I tested like this: 
* captured the RAW signal using T-Embed, and the decoded signal with Cardputer... This signal was sent by the Controller
* Replayed the RAW signal with T-Embed and saw the same decoded signal on the Cardputer and the gate was opening 
*

bmorcelli, 2024-12-30:
Depending on the system, if it has rolling code, there's a small risk of desynchronization of the key fob and the car.. 

Mostly RAW reading is a Bruce problem, that we need to figure out a way to do read and replay successfully..

Flipper Zero has tons of people developing for it, and a lot of resources (money, people amd hardware) to play with...and bruce, it the other hand, has only myself trying to make this raw reading work by messing with the RCSwitch library, with no money, no Flipper and no hackRf and low time 😔

thanatos84, 2025-01-07:
@bmorcelli I've tested just now to flash latest beta from the web flasher, tested to capture my garage door, saving and then opening and it works!

In the latests betas I received lot of noise when capturing and it didn't work, now without noise and working!

@rustysun9 , @eadmaster , @hunt3rbb666NotFound & @MayLaOwn  can you try also to doublecheck that works also for you?

bmorcelli, 2025-01-08:
@thanatos84 awesome 😎

thanatos84, 2025-01-10:
@hunt3rbb666NotFound I just tried with my garage door that works with portapack and flipper zero but didn't work with "bruce + m5stickc plus 2" and now it works capturing it and replying it, and also saving it and playing it

Today I've seen there's still be some noise and it captures some "fake" signals, I thought it was some interference in my house, but not, due to I've tried far away and still happened the same and appeared some (3-4) fake captured signals, but at least not the dozens that appeared in other previous versions 🙂

sk-oberon, 2025-01-19:
@bmorcelli I tested Firmware version 91d261c on M5StickC PLUS2 - RF record, save and replay (also from within the Files) works fine - both fixed and rolling code remotes.

Regarding noise/fake signals - In my other project that involves CC1101 module I had troubles with them as well and I managed to get rid of this issue by narrowing down the RF bandwidth to 60 kHz with ELECHOUSE_cc1101.setRxBW(60).

BTW - There are some glitches in beta FW (intentional maybe?), e.g.:
* when you hold down the ESC/return button (inside some app), "circle animation" doesn't appear and menu is being scrolled instead.
* when you hold down the ESC/return button (inside main menu), "pwr off in" is not in red color anymore.
* I have a feeling that navigation through the keyboard works better in stable - it kind of looks that debounce doesn't work very well

I have oscilloscope, RTL-SDR Blog V4 and many RF modules available - if there's something I can help with, I'm more than willing.

bmorcelli, 2025-01-19:
Nice . I'm happy that RF functions are working well..

The navigation difference is due to the way we are treating the inputs now.. we are reading them in a separate task, and then it gets a little harder to manage the long press logics, but the feedback is important, because this change had to be almost imperceptible

killerbe, 2025-01-19:
Hey, all.
Problem with SubGHZ. There is a call with a transmitter working on 433.92. The key is static, not dynamic. When saving, does not play the call, although when you just catch, you can play, but only when you save it does not work correctly, with barriers not sovsehm.
Tried to fix the code, the only thing prescribed the key and the sequence of bits transmitted.

Unfortunately I can attach a file with captured data before updating the code. With the new more readable did not save. But even saving raw data, does not play, I do not understand what's wrong. Can you tell me please?

Example 1:
Filetype: Bruce SubGhz File
Version 1
Frequency: 433657024
Preset: 0
Protocol: RAW
RAW_Data: 5489 -18 298 -7 203 -7 202 -7 231 -7 214 -7 221 -6 5571 -7 418 -7 203 -7 202 -7 221 -7 205 -7 5703 -18 453 -7 205 -7 210 -7 212 -7 206 -7 5661 -7 256 -7 205 -7 203 -7 220 -7 205 -7 5869 -7 300 -7 205 -7 202 -7 221 -7 209 -7 5821 -18 393 -7 205 -7 203 -7 220 -6 207 -7 5720 -7 411 -7 205 -7 203 -7 221 -6 206 -7 5713 -18 397 -7 205 -7 202 -7 221 -7 208 -7 5714 -7 388 -7 204 -7 203 -7 221 -7 207 -7 5735 -18 351 -7 205 -7 203 -7 218 -7 212 -7 5758 -7 320 -7 205 -7 202 -7 221 -7 210 -7 5800 -18 234 -7 204 -7 203 -7 219 -7 205 -7 5882 -7 433 -7 205 -7 203 -13 212 -7 206 -7 5693 -18 430 -7 205 -7 211 -7 212 -7 205 -7 5684 -7 405 -7 205 -7 202 -7 219 -7 207 -7 5720 -18 276 -7 205 -7 203 -7 220 -7 206 -7 5837 -7 375 -6 205 -7 203 -7 220 -7 210 -7 5746 -18 261 -7 204 -7 203 -7 220 -7 206 -7 5853 -7 303 -7 205 -7 203 -7 220 -7 206 -7 5821 -18 343 -7 205 -7 203 -7 220 -7 206 -7 5770 -7 256 -7 205 -7 202 -7 221 -7 205 -7 5869 -18 298 -7 205 -7 203 -6 221 -7 206 -7 215 -6 226 -9 12 -13 12 -12 12 -12 13 -5290 6 -430 7 -205 7 -208 7 -212 7 -206 7 -214 7 -5462 18 -336 7 -203 7 -203 7 -220 7 -207 7 -215 6 -5557 7 -275 7 -203 7 -202 7 -5295 6 -354 7 -205 7 -202 7 -217 7 -207 7 -5769 7 -230 7 -205 7 -202 7 -215 7 -206 7 -5900 19 -373 7 -205 7 -202 7 -217 7 -208 7 -5741 7 -419 7 -204 7 -203 7 -216 7 -207 7 -5709 18 -403 7 -205 7 -202 7 -215 7 -207 7 -215 6 -226 9 -12 12 -13 12 -12 12 -12 5191 -7 389 -7 205 -7 203 -7 217 -7 205 -7 215 -7 5502 -7 373 -7 202 -7 203 -7 217 -7 209 -6 215 -7 5533 -19 272 -7 203 -7 202 -7 217 -7 208 -7 215 -6 5623 -7 367 -7 203 -7 202 -7 217 -7 209 -7 215 -6 5539 -19 361 -7 203 -7 202 -7 217 -7 208 -7 215 -7 5533 -8 431 -7 203 -7 202 -15 208 -7 204 -7 215 -7 5479 -18 251 -6 7 -221 7 -5548 7 -313 7 -203 7 -202 7 -223 7 -215 7 -221 6 -5575 18 -411 7 -203 7 -203 7 -222 7 -211 6 -222 6 -5470 7 -371 7 -202 7 -203 7 -231

Example 2:
Filetype: Bruce SubGhz File
Version 1
Frequency: 433919968
Preset: 1
Protocol: RcSwitch
Bit: 24
Key: я і яя?і�
TE: 249

Example 3:
Filetype: Bruce SubGhz File
Version 1
Frequency: 433919968
Preset: 1
Protocol: RcSwitch
Bit: 24
Key: я і яя?і�
TE: 249

bmorcelli, 2025-01-19:
Sounds like the key is being saved wrong.. that is not the expected to format.. it must be something like "00 00 00 00 11 22 AA BB"

I'll check this out

sk-oberon, 2025-01-19:
@bmorcelli I looks like issue FW v1.8 has, the one that was fixed already (at least in beta).
I answered @killerbe in [issue 675](https://github.com/pr3y/Bruce/issues/675).

killerbe, 2025-01-19:
Thanks for helps

> Sounds like the key is being saved wrong.. that is not the expected to format.. it must be something like "00 00 00 00 11 22 AA BB"
> 
> I'll check this out

bmorcelli, 2025-01-19:
> @bmorcelli I looks like issue FW v1.8 has, the one that was fixed already (at least in beta).
> I answered @killerbe in [#675](https://github.com/pr3y/Bruce/issues/675).

Great!! Thank you!! Saved a few hours of debug and testing 😚