BruceDevices/firmware #611: Add BLE Keyboard and Custom QR Feature

nikyy2, 2024-12-25:
# Hey, I would like to suggest a new feature in BLE. It would be great if you could add a BLE keyboard that uses the same keyboard UI currently used when entering passwords for Wi-Fi or entering custom QR.  

Using Bruce as a Bluetooth keyboard !!!

# It would be cool to have a custom QR code feature that use litteFS. Here’s the idea:  
- Users to upload saved text files (e.g., `.txt`) to LittleFS.  
- In the QR code section, provide an option to choose one of the uploaded `.txt` files to generate a QR code.  
- For instance, a file containing `https://bruce.computer/` could be selected and converted into a QR code.  

This would help sharing links, social media accounts, much easier and faster without needing to retype them every time.  

**Best regards,**  
**nikyy2**

dghlkxyvdyxj, 2024-12-30:
it would maybe not work for the stick but for cardputer would be nice

bmorcelli, 2024-12-31:
> It would be great if you could add a BLE keyboard that connect to device (Iphone, Android, Window,...) so users can use it as a Bluetooth Keyboard.

It is available for Cardputer and T-Deck, in other devices would be way too hard to type

>It would be cool to have a custom QR code feature that use litteFS.

It is available on Beta, you download the file bruce.conf, and add the QR codes you want.

closed 2024-12-31 by bmorcelli