BruceDevices/firmware #614: Battery indicator for CYD

BeqaMiqa, 2024-12-26:
**Is your feature request related to a problem? Please describe.**

The indicator doesn't work for the cyd-2432s028 

**Describe the solution you'd like**

We can use ADS1115 (low-power, 16-bit analog-to-digital converter):

https://sl.aliexpress.ru/p?key=pzzrG43

 to precisely measure battery voltage, but we know we can not use pins 22 and 27, because other devices need these pins, so we can use the UART pins 1 and 3 to configure as I2C , they are conveniently placed next to VIN pin. 
I made a small demo using this code (parts of the code are irrelevant so please ignore it):
`#include <Wire.h>
#include <Adafruit_ADS1X15.h>
#include <SPI.h>
#include <TFT_eSPI.h>
#include <XPT2046_Touchscreen.h>

// Define ADS1115 I2C address
Adafruit_ADS1115 ads;

// Define RX and TX pins for I2C
#define I2C_SDA_PIN 3
#define I2C_SCL_PIN 1

#define SCREEN_WIDTH 320
#define SCREEN_HEIGHT 240

#define XPT2046_IRQ 36  
#define XPT2046_MOSI 32 
#define XPT2046_MISO 39 
#define XPT2046_CLK 25  
#define XPT2046_CS 33  

#define TOUCH_CS XPT2046_CS
#define TOUCH_IRQ XPT2046_IRQ
#define SPI_FREQUENCY 2000000 

// Initialize TFT and touchscreen
SPIClass touchscreenSPI = SPIClass(VSPI);
XPT2046_Touchscreen touchscreen(XPT2046_CS, XPT2046_IRQ);
TFT_eSPI tft = TFT_eSPI();

// Battery icon bitmaps
// 'empty-battery', 32x32px
const unsigned char epd_bitmap_empty_battery [] PROGMEM = {
	//Bitmap here 
};
// 'low-battery', 32x32px
const unsigned char epd_bitmap_low_battery [] PROGMEM = {
	//Bitmap here 
};
// 'half-battery', 32x32px
const unsigned char epd_bitmap_half_battery [] PROGMEM = {
	//Bitmap here 
};
// 'battery', 32x32px
const unsigned char epd_bitmap_battery [] PROGMEM = {
	//Bitmap here
};
// 'full-battery', 32x32px
const unsigned char epd_bitmap_full_battery [] PROGMEM = {
	//Bitmap here
};

void setup() {
    Serial.begin(115200); 
    
   
    tft.init();
    tft.setRotation(1);
    tft.fillScreen(TFT_BLACK);
    
    // Initialize I2C for ADS1115
    Wire.begin(I2C_SDA_PIN, I2C_SCL_PIN);
    delay(100);
    
    // Initialize ADS1115
    if (!ads.begin()) {
        tft.println("ADS1115 Failed!");
        while (1);
    }
    
   
    touchscreen.begin();
    touchscreen.setRotation(1);
    
    tft.setTextColor(TFT_WHITE, TFT_BLACK);
    tft.setTextSize(2);
}

void loop() {
    // Read voltage from ADS1115
    int16_t adcValue = ads.readADC_SingleEnded(0);
    float voltage = adcValue * 0.1875 / 1000.0; // Convert to voltage

    // Determine battery percentage and icon
    const unsigned char *batteryIcon;
    String batteryLevel;
    if (voltage <= 3.3) {
        batteryIcon = epd_bitmap_empty_battery;
        batteryLevel = "0-20%";
    } else if (voltage <= 3.48) {
        batteryIcon = epd_bitmap_low_battery;
        batteryLevel = "20-40%";
    } else if (voltage <= 3.66) {
        batteryIcon = epd_bitmap_half_battery;
        batteryLevel = "40-60%";
    } else if (voltage <= 3.84) {
        batteryIcon = epd_bitmap_battery;
        batteryLevel = "60-80%";
    } else if (voltage <= 4.02) {
        batteryIcon = epd_bitmap_full_battery;
        batteryLevel = "80-100%";
    } else {
        batteryIcon = epd_bitmap_full_battery;
        batteryLevel = "100%";
    }

    tft.fillScreen(TFT_BLACK);
    tft.setCursor(10, 50);
    tft.println("Battery Status:");
    tft.setCursor(10, 100);
    tft.printf("Voltage: %.2f V", voltage);
    tft.setCursor(10, 150);
    tft.printf("Level: %s", batteryLevel.c_str());
    tft.drawBitmap(200, 50, batteryIcon, 32, 32, TFT_WHITE);

    delay(1000); 
}`

https://github.com/user-attachments/assets/d74d2c6a-7130-4456-a800-93b93cb860bf

Powering the ADS1115 using the onboard 3.3V pin is possible, but this sets 3.3V as the reference voltage for its ADC readings. Since the ADS1115 can only measure voltages below its reference voltage, the LiPo battery voltage (up to ~4.2V) would exceed this limit. To safely measure the battery voltage in this setup, it’s necessary to step it down using a voltage divider. For example, a divider with two 10k resistors can reduce the battery voltage to a range that is within the ADS1115’s measurable limits.

Alternatively, using the LX-LCBST module eliminates the need for a voltage divider entirely. The module boosts the battery voltage to a regulated 5V output for the VIN pin and powers the ADS1115. Since the battery voltage will always be below 5V, it falls safely within the ADS1115’s measurement range, simplifying the circuit.

https://sl.aliexpress.ru/p?key=vLNrGUH

Additionally, powering the VIN pin directly from the battery might not be ideal. The ESP32 documentation typically recommends a regulated 5V input for the VIN pin. The LX-LCBST module ensures stable operation for both the ESP32 and the ADS1115.

**Describe alternatives you've considered**

**Additional context**

bmorcelli, 2024-12-26:
This is something you can make on your side.. it doesn't make any sense to add something that the board can't do by itself

BeqaMiqa, 2024-12-27:
I mean we are adding everything else for operations (cc1110, nrf24, ch3929) the cyd can't all those without an external board,might as well add an external adc, no? And also as a side node pins 1 and 3 are underutilized, would be cool to use them in combination with 22 and 27 with something maybe

BeqaMiqa, 2024-12-27:
With a 3 d printed case it will be seem less too

bmorcelli, 2024-12-30:
too much work for something irrelevant, if you want it so much, you can add it, it is open source.

closed 2024-12-30 by bmorcelli