BruceDevices/firmware #616: NTP time 1 hour behind (DST issue?)

obsidianxenon, 2024-12-27:
When I select the NTP timezone: Sydney, the time displayed is one hour behind.

**Steps To Reproduce:**
- Device used: M5Stack Cardputer
- Configure clock
- Set NTP timezone
- Select Sydney

**Expected behavior:**
The time should be correct.

**Actual behavior:**
The time is an hour behind.

**Additional context / logs:**
At time of writing, it is daylight savings, so that could have something to do with it.

ademCoin, 2024-12-27:
me too for Paris France

Lamnxzp, 2024-12-27:
> me too for Paris France

Works fine for me..

Lamnxzp, 2024-12-27:
@obsidianxenon Yes, it is a DST issue.

https://github.com/pr3y/Bruce/blob/aa560a239b56c1789230ffd08b9c8d2c662cdb59/src/core/settings.cpp#L316-L329
Here are the possible options for fixing the problem:
- Show "UTC+2, UTC+3, .., UTC+11" instead of "Sydney", "Paris"
- Use an API like http://worldtimeapi.org, to determine what timezone, "Sydney"  for example, is currently there.

Not sure which one is better.

obsidianxenon, 2024-12-27:
@Lamnxzp Ok I'll look at those. Just put of curiosity, does NTP not account for DST? If it doesn't, why not? If it does, are there any plans to bring DST to Bruce? Thanks.

Lamnxzp, 2024-12-27:
@obsidianxenon
Currently in Bruce, If you select Sydney it will just set your timezone at `10`, so UTC+10.
This will cause problems during DST, yes..

rouing, 2024-12-28:
@bmorcelli Assign this one to me.
@obsidianxenon We dont support DST at the moment, but it is something that should be simple to do. Ill get around to this.

bmorcelli, 2024-12-28:
We were talking about this issue today, to use GMT-x instead of Country names.. so menu was going from -12 to +12 or something like that..

And theres an issue with places with half timezones, like India and Australia... Like GMT-x.5

rouing, 2024-12-28:
https://github.com/JChristensen/Timezone/blob/master/src/Timezone.h

It appears there is a means to setup Timechange rules and such. Ill keep looking into this.

obsidianxenon, 2024-12-28:
@rouing Ok thanks mate 👍

lahirunirmalx, 2024-12-31:
> We were talking about this issue today, to use GMT-x instead of Country names.. so menu was going from -12 to +12 or something like that..
> 
> And theres an issue with places with half timezones, like India and Australia... Like GMT-x.5

I already open Issue for this . didn't had time to resolve this   https://github.com/pr3y/Bruce/issues/543

obsidianxenon, 2024-12-31:
@bmorcelli There are also places that don't have DST, like Brisbane. There may have to be a location thing as well.

bmorcelli, 2024-12-31:
Brazil doesn't have too.. it might be easier to setup the clock Manually

obsidianxenon, 2024-12-31:
@bmorcelli I guess you're correct. I was just looking for an easier way rather than having to set it every time I power on.

rouing, 2024-12-31:
https://github.com/rouing/Bruce/tree/dev-timezones

Im sort of working on this now. I realized were all over the place with the time control so its gonna need a refactor. This will take a while.