BruceDevices/firmware #622: NRF24 Stronger Transmission

probaskid, 2024-12-27:
The NRF24 code is quite weak , there should be an option at which strength the user desires (ranging from weak to medium to high) , for the high function we could replace the current jammer.cpp code to something like 

#include "nrf_jammer.h"
#include <RF24.h>
#include "core/globals.h"
#include "core/mykeyboard.h"
#include "core/display.h"

// Declare the RF24 object that is used in the jammer code
RF24 radio(NRF24_CE_PIN, NRF24_SS_PIN);  // Assuming NRF24 pins are defined somewhere

// Function to start the jammer (calls the actual jamming logic)
void nrf_jammer() {
    nrf_jammer_max_power();  // Calls the jammer function
}

// Function to initialize the NRF24 module
bool nrf_start() {
    return radio.begin();  // Initializes the radio and returns true if successful
}

// Jammer code, remains unchanged as requested
void nrf_jammer_max_power() {
  #if defined(NRF24_CE_PIN) && defined(NRF24_SS_PIN) && defined(USE_NRF24_VIA_SPI)
    // Initialize NRF24L01+ module (already declared in the global scope as `radio`)

    // Full range of channels in the 2.4 GHz ISM band (0-125)
    byte hopping_channel[] = {0, 2, 4, 6, 8, 10, 12, 14, 16, 18, 20, 22, 24, 26, 28, 30, 
                              32, 34, 36, 38, 40, 42, 44, 46, 48, 50, 52, 54, 56, 58, 60, 
                              62, 64, 66, 68, 70, 72, 74, 76, 78, 80, 82, 84, 86, 88, 90, 
                              92, 94, 96, 98, 100, 102, 104, 106, 108, 110, 112, 114, 116, 
                              118, 120, 122, 124};
    byte ptr_hop = 0;  // Pointer for channel hopping

    if (nrf_start()) {
        Serial.println("NRF24 initialized successfully");

        // Set maximum power output
        radio.setPALevel(RF24_PA_MAX);  // Maximum power level
        radio.setDataRate(RF24_2MBPS);  // Maximum data rate
        radio.startConstCarrier(RF24_PA_MAX, 127);  // Constant carrier at full power (max amplitude)
        radio.setAddressWidth(3);  // Optional for stability
        radio.setPayloadSize(2);  // Optional for reliability

        drawMainBorder();
        tft.setCursor(10, 28);
        tft.setTextSize(FM);
        tft.println("MAX POWER BLE Jammer:");
        tft.setCursor(10, tft.getCursorY() + 8);
        tft.println("Press button to stop!");

        delay(200);

        // Randomize initial channel hopping point for unpredictability
        ptr_hop = random(0, sizeof(hopping_channel));

        // Aggressive channel hopping loop
        while (!checkSelPress()) {
            ptr_hop++;  // Move to next channel
            if (ptr_hop >= sizeof(hopping_channel)) {
                ptr_hop = 0;  // Loop back to the start
            }
            
            // Hop to the next channel as quickly as possible
            radio.setChannel(hopping_channel[ptr_hop]);

            // Minimum delay for aggressive hopping
            delay(5);  // 5ms delay between hops
        }

        // Stop jamming when the button is pressed
        radio.stopConstCarrier();  // Halt constant carrier transmission
        // radio.powerDown();  // Optional: Fully power down the module
    } else {
        Serial.println("Failed to initialize NRF24");
        displayError("NRF24 not detected");
        delay(500);
    }
  #endif
}
This would use up more power but it would be much stronger as 1. It has a wider channel hopping range , 2. It hops between channels more aggressively , quickly and unpredictably.
However it would use up more power and for max results a stable power and durability source.

rouing, 2024-12-29:
Isnt this basically what the code already is?

probaskid, 2024-12-29:
> Isnt this basically what the code already is?
 Nope , this covers wider range of channels and hops channels very aggressively to avoid defence mechanism by devices

rouing, 2024-12-29:
Its literally the same code, slightly different spacing. This is not "more powerful" The only change here is added channels which is fine but id rather allow the user to set the channels they want. The current ones target Bluetooth specifically.
Current Code
[image]

Compared - The same:
[image]

Compared - The same:
[image]

rouing, 2024-12-29:
[image]
And the 127 here is the channel you start on, not a power level.

closed 2024-12-30 by bmorcelli