BruceDevices/firmware #654: T-embed cc1101 bruce 1.8 does not send rf in 868mhz range

antoniojlara, 2025-01-05:
First of all, thank you for the great work.

 In my t-embed cc1101 in the Rf section it copies and sends the 440hz frequencies well, but in 868hz it only copies them at the time of transmission, it is not detected by the monitoring from my cardputer nor is it detected by the control unit which is a fixed code.

bmorcelli, 2025-01-06:
Depending on the CC1101 you have on Cardputer, if it is labeled 433, it won't perform good receiving other frequency ranges (300-350 and 760-930) because the antenna circuit doesn't have the right setup..  T-Embed, in the other hand, has a antenna circuit that is set by software.. so it performs good at all ranges of frequency.

antoniojlara, 2025-01-06:
Hello! Thanks, on cardputer I have cc11011 v2 433MHz/868MHz it also copies them but does not send well, or at least with my 868mhz port the same thing happens in t-embed.

bmorcelli, 2025-01-20:
Can you test the last release?

TheCherryTomato, 2025-01-20:
868mhz works in 1.8.1. Can confirm.

closed 2025-01-20 by bmorcelli