BruceDevices/firmware #656: Possibility to rename bluetooth name

swesws, 2025-01-06:
some windows systems do not see bluetooth device with a name like "keyboard_**".
using Arduino ide i wrote my own firmware that creates a Bluetooth connection with the same name as my phone, after which Windows began to see my device in the list of Bluetooth connections.
function to rename the bluetooth name could solve this problem.

my device: m5stickC plus2

eddietheyeti1126, 2025-02-27:
can you share that frimware i am also trying to rename my ble device name for bruce.

Lamnxzp, 2025-02-27:
@eddietheyeti1126 @swesws
This feature has been added here: #765 

It's available on Bruce Beta (https://bruce.computer/flasher -> BetaRelease).
It's in Config -> Network Creds -> BLE Name

IncursioHack, 2025-03-21:
Validated, thanks for the contribution @Lamnxzp

closed 2025-03-21 by IncursioHack