BruceDevices/firmware #660: Support ESP32-2432S024R 2.4" Resistive Touch USB Type-C Only

Ruslan228asr, 2025-01-08:
Good day! My name is Ruslan, and I'm writing from Ukraine. Recently, I encountered a problem with the firmware for the CYD-2432S028R (Type-C). The standard CYD-2432S028 firmware was installed on it, but there was no image. However, the CYD-2432W328C firmware displayed the image, but the touch screen didn't work.

I tried installing all the firmware versions from the official website, but nothing worked. The display itself is functional, as I've used it before. I also confirmed its operability with the Marauder firmware, where everything works fine.

I hope you can add support for this board. Many users on forums and Discord channels have reported similar issues with this board.

bmorcelli, 2025-01-08:
We don't have this device to setup the touchscreen and screen, so it's difficult to do things blindly..

Do you have repository examples of codes that work for this display? So we can get information and implement it?

RandomDHR6, 2025-01-08:
i think this one would work : https://github.com/rzeldent/esp32-smartdisplay-demo

Ruslan228asr, 2025-01-09:
> i think this one would work : https://github.com/rzeldent/esp32-smartdisplay-demo

I think this one will also work

Ruslan228asr, 2025-01-09:
I have a beta version installed for CYD-2432S024c, and CYD-2432W28. But all had two problems. More precisely: the touchscreen does not work, and the animations do not work correctly (I mean the bruce animation)

bmorcelli, 2025-01-09:
> The touchscreen does not work,

Different board, different touch pins/drive

 > and the animations do not work correctly (I mean the bruce animation) 

If you're running the Launcher Version, it doesn't have the animation

Ruslan228asr, 2025-01-09:
> > The touchscreen does not work,
> 
> Different board, different touch pins/drive
> 
>  > and the animations do not work correctly (I mean the bruce animation) 
> 
> If you're running the Launcher Version, it doesn't have the animation
> 

yes, I understand

Ruslan228asr, 2025-01-10:
will it be possible to support CYD-2432S024r?

Neo2a, 2025-01-15:
Touch function and battery level indicator is working well on this firmware. https://github.com/Spooks4576/Ghost_ESP
I installed it to ESP32-2432S024R board. Maybe possible to migrate setups of touch function.

bmorcelli, 2025-01-15:
> Touch function and battery level indicator is working well on this firmware. https://github.com/Spooks4576/Ghost_ESP I installed it to ESP32-2432S024R board. Maybe possible to migrate setups of touch function.

I sent a patch for it now, will be available in about 15 minutes to install

Battery level indicator for these CYD boards rely on a separate device connected to the GPIO.. we (Bruce devs) prefere to use these GPIO for RF, RFID, GPS and more interesting stuff than showing a percentage of battery on screen.. 

so, if it comes for us to decide to add or not, the answer is "its not gonna happen"..

Ruslan228asr, 2025-01-15:
The touchscreen doesn't work. But I can send you the logs, maybe they will help in some way.
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (1347) esp_core_dump_flash: No core dump partition x櫹��
E (1347) esp_core_dump_flash: No core dump partition found!
E (1351) esp_ota_ops: not found otadata
[    21][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    48][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    57][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    68][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
[    83][D][main.cpp:302] setup(): Total heap: 195864
[    88][D][main.cpp:303] setup(): Free heap: 160812
[    93][D][main.cpp:306] setup(): PSRAM Not Found
[    97][D][main.cpp:307] setup(): Total PSRAM: 0
[   102][D][main.cpp:308] setup(): Free PSRAM: 0
[   108][I][interface.cpp:35] _setup_gpio(): Touch IC Started
[   421][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x401cda18 arg=0x3ffc0970
[   483][I][config.cpp:176] fromFile(): Using config from file
[   513][I][config.cpp:176] fromFile(): Using config from file
[   551][I][interface.cpp:70] _setBrightness(): dutyCycle for bright 0-255: 255
E (1890) ledc: ledc_set_duty(725): LEDC is not initialized
E (1891) ledc: ledc_update_duty(653): LEDC is not initialized
{
  "priColor": "a80f",
  "secColor": "880f",
  "bgColor": "0",
  "rot": 1,
  "dimmerSet": 10,
  "bright": 100,
  "tmz": 0,
  "soundEnabled": 1,
  "wifiAtStartup": 0,
  "ledBright": 75,
  "ledColor": "0",
  "webUI": {
    "user": "admin",
    "pwd": "bruce"
  },
  "wifiAp": {
    "ssid": "BruceNet",
    "pwd": "brucenet"
  },
  "wifi": {},
  "irTx": 22,
  "irRx": 22,
  "rfTx": 27,
  "rfRx": 22,
  "rfModule": 0,
  "rfFreq": 433.92,
  "rfFxdFreq": 1,
  "rfScanRange": 3,
  "rfidModule": 0,
  "mifareKeys": [],
  "gpsBaudrate": 9600,
  "startupApp": "",
  "wigleBasicToken": "",
  "devMode": 0,
  "disabledMenus": [],
  "qrCodes": [
    {
      "menuName": "Bruce AP",
      "content": "WIFI:T:WPA;S:BruceNet;P:brucenet;;"
    },
    {
      "menuName": "Bruce Wiki",
      "content": "https://github.com/pr3y/Bruce/wiki"
    },
    {
      "menuName": "Bruce Site",
      "content": "https://bruce.computer"
    },
    {
      "menuName": "Rickroll",
      "content": "https://youtu.be/dQw4w9WgXcQ"
    }
  ]
}[   680][I][config.cpp:195] saveFile(): config file written successfully
[   698][W][LittleFS.cpp:77] begin(): LittleFS Already Mounted!
[   956][E][vfs_api.cpp:105] open(): /littlefs/boot.jpg does not exist, no permits for creation
[  1303][I][config.cpp:176] fromFile(): Using config from file
[ 11247][I][interface.cpp:70] _setBrightness(): dutyCycle for bright 0-255: 12
[ 12274][I][interface.cpp:70] _setBrightness(): dutyCycle for bright 0-255: 255

H4NT3R0, 2025-01-16:
[image]

Neo2a, 2025-01-16:
The ghost ESP is working. Can you implement that setups into your firmware? https://github.com/Spooks4576/Ghost_ESP

https://github.com/user-attachments/assets/463f44e6-a7fe-4f27-9eee-3746c07de92c

bmorcelli, 2025-01-16:
Ghost uses lvgl and esp-idf for his project.. his libs doesn't work on our environment.. there's an easy way out for this problem.. I'll try it out in a few hours

Neo2a, 2025-01-16:
Thank you for yor effort.

Ruslan228asr, 2025-01-17:
> Ghost uses lvgl and esp-idf for his project.. his libs doesn't work on our environment.. there's an easy way out for this problem.. I'll try it out in a few hours 

ok, thanks for supporting this device. I have a question. Should I send you the logs during testing?

bmorcelli, 2025-01-17:
It is a good practice, as I don't have this device

RandomDHR6, 2025-01-17:
I would test it too but the flex tape that connects the touch screen to the esp has a little cut and it doesn't work unfortunately.

bmorcelli, 2025-01-17:
Can you try installin the Beta from my repo??

Flasher is here https://bmorcelli.github.io/Bruce/flasher

Ruslan228asr, 2025-01-17:
> Can you try installin the Beta from my repo??
> 
> Flasher is here https://bmorcelli.github.io/Bruce/flasher

there are problems with loading here is a screenshot: [image]

I'll send the logs now

Ruslan228asr, 2025-01-17:
> > Can you try installin the Beta from my repo??
> > Flasher is here https://bmorcelli.github.io/Bruce/flasher
> 
> there are problems with loading here is a screenshot: [image]
> 
> I'll send the logs now

ndler
E (12101) task_wdt: CPU 1: loopTask
E (12101) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401E32B6:0x3FFC1BBC |<-CORRUPTED

E (12101) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400849E5:0x3FFC23EC |<-CORRUPTED

E (17101) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (17101) task_wdt:  - IDLE0 (CPU 0)
E (17101) task_wdt: Tasks currently running:
E (17101) task_wdt: CPU 0: InputHandler
E (17101) task_wdt: CPU 1: loopTask
E (17101) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401E32B6:0x3FFC1BBC |<-CORRUPTED

E (17101) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400849E5:0x3FFC23EC |<-CORRUPTED

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (1354) esp_core_dump_flash: No core dump partition f�f~nd!
E (1354) esp_core_dump_flash: No core dump partition found!
E (1359) esp_ota_ops: not found otadata
[    21][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    47][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    57][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    68][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
[    83][D][main.cpp:298] setup(): Total heap: 195872
[    88][D][main.cpp:299] setup(): Free heap: 160760
[    92][D][main.cpp:302] setup(): PSRAM Not Found
[    97][D][main.cpp:303] setup(): Total PSRAM: 0
[   102][D][main.cpp:304] setup(): Free PSRAM: 0
[   415][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x401d1710 arg=0x3ffc0958
[   477][I][config.cpp:176] fromFile(): Using config from file
[   508][I][config.cpp:176] fromFile(): Using config from file
[   545][I][interface.cpp:79] _setBrightness(): dutyCycle for bright 0-255: 255
E (1893) ledc: ledc_set_duty(725): LEDC is not initialized
E (1894) ledc: ledc_update_duty(653): LEDC is not initialized
{
  "priColor": "a80f",
  "secColor": "880f",
  "bgColor": "0",
  "rot": 1,
  "dimmerSet": 10,
  "bright": 100,
  "tmz": 0,
  "soundEnabled": 1,
  "wifiAtStartup": 0,
  "ledBright": 75,
  "ledColor": "0",
[... 96 lines cut ...]
E (27088) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (27088) task_wdt:  - IDLE0 (CPU 0)
E (27088) task_wdt: Tasks currently running:
E (27088) task_wdt: CPU 0: InputHandler
E (27088) task_wdt: CPU 1: loopTask
E (27088) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401E32B6:0x3FFC1BBC |<-CORRUPTED

E (27088) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400849E5:0x3FFC23EC |<-CORRUPTED

E (32088) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (32088) task_wdt:  - IDLE0 (CPU 0)
E (32088) task_wdt: Tasks currently running:
E (32088) task_wdt: CPU 0: InputHandler
E (32088) task_wdt: CPU 1: loopTask
E (32088) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401E32B6:0x3FFC1BBC |<-CORRUPTED

E (32088) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400849E5:0x3FFC23EC |<-CORRUPTED

bmorcelli, 2025-01-17:
ok, updated again... you can try again in https://bmorcelli.github.io/Bruce/flasher

Ruslan228asr, 2025-01-17:
> ok, updated again... you can try again in https://bmorcelli.github.io/Bruce/flasher

you added color but bruce is not loading  
[image]

Ruslan228asr, 2025-01-17:
> > ок, обновил еще раз... можете попробовать еще раз в https://bmorcelli.github.io/Bruce/flasher
> 
> вы добавили цвет, но Брюс не загружается [image]

logs: 

E (12103) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (12103) task_wdt:  - IDLE0 (CPU 0)
E (12103) task_wdt: Tasks currently running:
E (12103) task_wdt: CPU 0: InputHandler
E (12103) task_wdt: CPU 1: loopTask
E (12103) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401E329E:0x3FFC1BBC |<-CORRUPTED

E (12103) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400849E5:0x3FFC23EC |<-CORRUPTED

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (1354) esp_core_dump_flash: No core dump partition f��ֹ��
E (1354) esp_core_dump_flash: No core dump partition found!
E (1359) esp_ota_ops: not found otadata
[    21][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    47][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    57][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    68][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
[    82][D][main.cpp:298] setup(): Total heap: 195864
[    87][D][main.cpp:299] setup(): Free heap: 160752
[    92][D][main.cpp:302] setup(): PSRAM Not Found
[    97][D][main.cpp:303] setup(): Total PSRAM: 0
[   101][D][main.cpp:304] setup(): Free PSRAM: 0
[   416][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x401d16f8 arg=0x3ffc0958
[   478][I][config.cpp:176] fromFile(): Using config from file
[   509][I][config.cpp:176] fromFile(): Using config from file
[   546][I][interface.cpp:79] _setBrightness(): dutyCycle for bright 0-255: 255
E (1893) ledc: ledc_set_duty(725): LEDC is not initialized
E (1894) ledc: ledc_update_duty(653): LEDC is not initialized
{
  "priColor": "a80f",
  "secColor": "880f",
  "bgColor": "0",
  "rot": 1,
  "dimmerSet": 10,
  "bright": 100,
  "tmz": 0,
  "soundEnabled": 1,
  "wifiAtStartup": 0,
  "ledBright": 75,
  "ledColor": "0",
  "webUI": {
    "user": "admin",
    "pwd": "bruce"
  },
  "wifiAp": {
    "ssid": "BruceNet",
    "pwd": "brucenet"
  },
  "wifi": {},
  "irTx": 22,
  "irRx": 22,
  "rfTx": 27,
  "rfRx": 22,
  "rfModule": 0,
  "rfFreq": 433.92,
  "rfFxdFreq": 1,
  "rfScanRange": 3,
  "rfidModule": 0,
  "mifareKeys": [],
  "gpsBaudrate": 9600,
  "startupApp": "",
  "wigleBasicToken": "",
  "devMode": 0,
  "disabledMenus": [],
  "qrCodes": [
    {
      "menuName": "Bruce AP",
      "content": "WIFI:T:WPA;S:BruceNet;P:brucenet;;"
    },
    {
      "menuName": "Bruce Wiki",
      "content": "https://github.com/pr3y/Bruce/wiki"
    },
    {
      "menuName": "Bruce Site",
      "content": "https://bruce.computer"
    },
    {
      "menuName": "Rickroll",
      "content": "https://youtu.be/dQw4w9WgXcQ"
    }
  ]
}[   686][I][config.cpp:195] saveFile(): config file written successfully
[   704][W][LittleFS.cpp:77] begin(): LittleFS Already Mounted!
E (7109) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (7109) task_wdt:  - IDLE0 (CPU 0)
E (7109) task_wdt: Tasks currently running:
E (7109) task_wdt: CPU 0: InputHandler
E (7109) task_wdt: CPU 1: loopTask
E (7109) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401E329E:0x3FFC1BBC |<-CORRUPTED

E (7109) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400849E5:0x3FFC23EC |<-CORRUPTED

E (12109) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (12109) task_wdt:  - IDLE0 (CPU 0)
E (12109) task_wdt: Tasks currently running:
E (12109) task_wdt: CPU 0: InputHandler
E (12109) task_wdt: CPU 1: loopTask
E (12109) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401E329E:0x3FFC1BBC |<-CORRUPTED

E (12109) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400849E5:0x3FFC23EC |<-CORRUPTED

E (17109) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (17109) task_wdt:  - IDLE0 (CPU 0)
E (17109) task_wdt: Tasks currently running:
E (17109) task_wdt: CPU 0: InputHandler
E (17109) task_wdt: CPU 1: loopTask
E (17109) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401E329E:0x3FFC1BBC |<-CORRUPTED

E (17109) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400849E5:0x3FFC23EC |<-CORRUPTED

bmorcelli, 2025-01-17:
I tried a last trick, feeding the watchdog.. you can try again

Ruslan228asr, 2025-01-17:
> I tried a last trick, feeding the watchdog.. you can try again 

nothing has changed, it hangs on loading. But when pressing "reset" the inscription can be complete or vice versa.

Ruslan228asr, 2025-01-17:
> Thank you for yor effort.
could you check the firmware?

bmorcelli, 2025-01-17:
I tried using the touch driver from TFT_eSPI, but didn't work..

I will make changes on our cyd touch lib to use the shared bus when I have time

bmorcelli, 2025-01-18:
New try on https://bmorcelli.github.io/Bruce/flasher

H4NT3R0, 2025-01-18:
[image]

Neo2a, 2025-01-18:
I installed the latest version. The screen color is inverted. 
And unfortunately the touch function does not work.

Ruslan228asr, 2025-01-18:
> [image]

Try beta version

RandomDHR6, 2025-01-18:
just relaod the page

Ruslan228asr, 2025-01-18:
> New try on https://bmorcelli.github.io/Bruce/flasher

touchscreen doesn't work, evrerything is loading. I send logs:

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (1354) esp_core_dump_flash: No core dump partition fo��.Vj
E (1354) esp_core_dump_flash: No core dump partition found!
E (1359) esp_ota_ops: not found otadata
[    21][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    48][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    57][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    68][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
[    83][D][main.cpp:298] setup(): Total heap: 195832
[    88][D][main.cpp:299] setup(): Free heap: 160720
[    93][D][main.cpp:302] setup(): PSRAM Not Found
[    97][D][main.cpp:303] setup(): Total PSRAM: 0
[   102][D][main.cpp:304] setup(): Free PSRAM: 0
[   421][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x401d16d0 arg=0x3ffc0968
[   442][W][sd_diskio.cpp:174] sdCommand(): no token received
[   549][W][sd_diskio.cpp:174] sdCommand(): no token received
[   655][W][sd_diskio.cpp:174] sdCommand(): no token received
[   760][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[   767][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[   773][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[   782][W][sd_diskio.cpp:174] sdCommand(): no token received
[   888][W][sd_diskio.cpp:174] sdCommand(): no token received
[   994][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1100][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  1119][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1225][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1331][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1436][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  1442][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  1449][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1458][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1563][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1669][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1775][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  1850][I][config.cpp:176] fromFile(): Using config from file
[  1869][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1976][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2082][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2187][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  2194][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  2200][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  2209][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2314][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2420][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2526][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  2601][I][config.cpp:176] fromFile(): Using config from file
[  2638][I][interface.cpp:85] _setBrightness(): dutyCycle for bright 0-255: 255
E (3986) ledc: ledc_set_duty(725): LEDC is not initialized
E (3987) ledc: ledc_update_duty(653): LEDC is not initialized
{
  "priColor": "a80f",
  "secColor": "880f",
  "bgColor": "0",
  "rot": 1,
  "dimmerSet": 10,
  "bright": 100,
  "tmz": 0,
[... 49 lines cut ...]
[  3027][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3132][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  3139][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  3145][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  3154][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3259][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3365][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3471][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  3480][I][interface.cpp:67] _post_setup_gpio(): Touch IC Started
[  3495][E][vfs_api.cpp:113] exists(): File system is not mounted
[  3505][E][vfs_api.cpp:105] open(): /littlefs/boot.jpg does not exist, no permits for creation
[  3514][E][vfs_api.cpp:113] exists(): File system is not mounted
[  3523][E][vfs_api.cpp:105] open(): /littlefs/boot.gif does not exist, no permits for creation
[  3828][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3935][W][sd_diskio.cpp:174] sdCommand(): no token received
[  4041][W][sd_diskio.cpp:174] sdCommand(): no token received
[  4146][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  4153][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  4159][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  4168][W][sd_diskio.cpp:174] sdCommand(): no token received
[  4274][W][sd_diskio.cpp:174] sdCommand(): no token received
[  4380][W][sd_diskio.cpp:174] sdCommand(): no token received
[  4486][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  4566][I][config.cpp:176] fromFile(): Using config from file
>

teuchezh, 2025-01-19:
Can you add support of ESP32-2432S024R with capacitive touch? Touch controller is CST820

bmorcelli, 2025-01-19:
> Can you add support of ESP32-2432S024R with capacitive touch? Touch controller is CST820

I believe this one is already ported in this option "CYD-2432W328C(inv_colors)_and_CYD-2432S024C"

bmorcelli, 2025-01-20:
New try on https://bmorcelli.github.io/Bruce/flasher

I need Serial Logs when touching screen..

Ruslan228asr, 2025-01-20:
> > Новая попытка на https://bmorcelli.github.io/Bruce/flasher
> > 
> > Мне нужны последовательные журналы при касании экрана.
> 
> [esp-web-tools-logs (1).txt](https://github.com/user-attachments/files/18479804/esp-web-tools-logs.1.txt)

I hope this gave you something, because when you press the touchscreen, nothing is written in the logs.

bmorcelli, 2025-01-20:
Nice... now I am very comfident that it will work:

 try on https://bmorcelli.github.io/Bruce/flasher (Beta Release)

teuchezh, 2025-01-20:
> I believe this one is already ported in this option "CYD-2432W328C(inv_colors)_and_CYD-2432S024C"

Tested this version, and looks like working not bad. Need more time for explore stability

Ruslan228asr, 2025-01-20:
> Nice... now I am very comfident that it will work:
> 
> try on https://bmorcelli.github.io/Bruce/flasher (Beta Release)

touchscreen doesn't work still the same

[esp-web-tools-logs (1).txt](https://github.com/user-attachments/files/18481357/esp-web-tools-logs.1.txt)

bmorcelli, 2025-01-21:
New Commit on https://bmorcelli.github.io/Bruce/flasher (Beta Release)

Ruslan228asr, 2025-01-21:
> New Commit on https://bmorcelli.github.io/Bruce/flasher (Beta Release)

no boot 

[esp-web-tools-logs (2).txt](https://github.com/user-attachments/files/18493922/esp-web-tools-logs.2.txt)

bmorcelli, 2025-01-21:
I love these logs..

Pushed one more commit, feeding the watchdog.. hope it works

Ruslan228asr, 2025-01-21:
> I love these logs..
> 
> Pushed one more commit, feeding the watchdog.. hope it works

ok i send logs 

[esp-web-tools-logs (3).txt](https://github.com/user-attachments/files/18495871/esp-web-tools-logs.3.txt)

Ruslan228asr, 2025-01-21:
> I love these logs..
> 
> Pushed one more commit, feeding the watchdog.. hope it works

the problem persists

bmorcelli, 2025-01-22:
New Commit on https://bmorcelli.github.io/Bruce/flasher (Beta Release)

Now I fully disabled the watchdog (or tried to)

Waiting for logs

Ruslan228asr, 2025-01-22:
> New Commit on https://bmorcelli.github.io/Bruce/flasher (Beta Release)
> 
> Now I fully disabled the watchdog (or tried to)
> 
> Waiting for logs
> 

Ok, I'll test it when I'm home.

Ruslan228asr, 2025-01-22:
> New Commit on https://bmorcelli.github.io/Bruce/flasher (Beta Release)
> 
> Now I fully disabled the watchdog (or tried to)
> 
> Waiting for logs
it restarts every time

[esp-web-tools-logs (4).txt](https://github.com/user-attachments/files/18507941/esp-web-tools-logs.4.txt)

ciccirix, 2025-01-25:
any news for 2.4 r??

Ruslan228asr, 2025-01-28:
> New try on https://bmorcelli.github.io/Bruce/flasher
 touch screen not working 

[esp-web-tools-logs (5).txt](https://github.com/user-attachments/files/18579310/esp-web-tools-logs.5.txt)

Ruslan228asr, 2025-01-28:
> > New try on https://bmorcelli.github.io/Bruce/flasher
>  touch screen not working 
> 
> [esp-web-tools-logs (5).txt](https://github.com/user-attachments/files/18579310/esp-web-tools-logs.5.txt)
> 

Stop, when going into sleep mode (when the screen is off) when you press the screen it turns it on. But I can't interact

bmorcelli, 2025-01-29:
great.. interrupt is working...

I just changed the Calibration data and Z threshold (left it more sensitive)..

as you can see in [this line](https://github.com/bmorcelli/Bruce/blob/cd197b6253666ff7c4bfdf1e147574354a7d2c7c/boards/CYD-2432S028/interface.cpp#L135), once you touch the screen it reads the data from the 

(wait 10 minutes from this post to try, please)

Ruslan228asr, 2025-01-29:
> great.. interrupt is working...
>
> I just changed the Calibration data and Z threshold (left it more sensitive)..
>
> as you can see in [this [...]

I installed it, but the sensor still doesn't work. I didn't have time to send the logs, I'll send everything today

Ruslan228asr, 2025-01-29:
> I installed it, but the sensor still doesn't work. I didn't have time to send the logs, I'll send everything today

i send logs 

[esp-web-tools-logs (6).txt](https://github.com/user-attachments/files/18590151/esp-web-tools-logs.6.txt)

ciccirix, 2025-01-29:
NEW VERSION HAS COLOR INVERTED BUT NOT WORK

bmorcelli, 2025-02-12:
Not yet.. I bought one of these CYDs to investigate

ciccirix, 2025-02-12:
👌

bmorcelli, 2025-02-15:
Hey,

Can you test here: https://buymeacoffee.com/bmorcelliz 
choose BetaRelease -> WIP -> CYD-2432W328R or 2432S024R"

lkmsky, 2025-02-15:
Everything works, thank you very much 

> Hey,
> 
> Can you test here: https://buymeacoffee.com/bmorcelliz choose BetaRelease -> WIP -> CYD-2432W328R or 2432S024R"

bmorcelli, 2025-02-17:
I added a Calibration process at the start of Bruxe, or this device, maybe it works now..

this calibration will be done one time, and to do it again you need to delete the "calData" file from LittleFs and restart..

can you test it, please?
https://bmorcelli.github.io/Bruce/flasher

BetaRelease -> WIP -> CYD-2432W328R or 2432S024R"

[9 more replies]

closed 2025-02-17 by bmorcelli