BruceDevices/firmware #669: Volume doesn't work with bad BLE

actuallybytes, 2025-01-10:
**Describe the bug:**
Using "VOLUME UP" in bad BLE doesn't seem to work.

**Steps To Reproduce:**
Settings the volume up with bad BLE. (If anyone knows how to instantly put it at max I would really would want to know that too.)
- Device used: 
- M5StickC Plus2

**Expected behavior:**
I expected the volume to go up.

**Actual behavior:**
Instead, it ran as a string.

dghlkxyvdyxj, 2025-01-13:
its not an issue with bruce (i think) if you use the bad ble on windows you can make it open powershell and it will set the volume on max
[rickroll pc.txt](https://github.com/user-attachments/files/18397569/rickroll.pc.txt) here an example

actuallybytes, 2025-01-13:
could you just send the text here

closed 2025-01-16 by actuallybytes