BruceDevices/firmware #671: Deauthenticaion problems on CYD

7wp81x, 2025-01-11:
Wifi Deauthentication doesn't work in raw sniffer, brucenogotchi on CYD, I tried all deauth features on it but two of them works "deauth flood" and "Deauth" under "Wifi atks > Target atks" including "Deauth+clone", also the "Station deauth" on "Scan hosts" crashes the CYD maybe this because that the CYD was connected to the network while also sending deauth frames?

I am using CYD that has microUSB, and USB-C, (flashed with firmware that has 2-USB name)

actuallybytes, 2025-01-11:
Is the network 2.4 GHz or 5 GHz?

7wp81x, 2025-01-12:
> Is the network 2.4 GHz or 5 GHz?

well it won't work in 5 GHz cuz esp32 is only 2.4GHz... but yeah I tried it on open hotspot and with WPA/WPA2 network i did also try to use tenda and tplink router but still not working..

but the deauth while sniffing seems to work in this project: https://github.com/risinek/esp32-wifi-penetration-tool

7wp81x, 2025-01-12:
> > Is the network 2.4 GHz or 5 GHz?
> 
> well it won't work in 5 GHz cuz esp32 is only 2.4GHz... but yeah I tried it on open hotspot and with WPA/WPA2 network i did also try to use tenda and tplink router but still not working..
> 
> but the "deauthing" while sniffing seems to work in this project: https://github.com/risinek/esp32-wifi-penetration-tool

i tried that project on network

actuallybytes, 2025-01-12:
I have the same device. Tommorow I'll try to see if it works on my CYD. If yes, then we'll look into your CYD, if not, it's probably an issue with the Bruce firmware, or my CYD.

actuallybytes, 2025-01-13:
I can't really help you because my CYD doesn't even see my hotspot, even when it's close to my phone. I can't do this on my home network because it's 5 GHz.

closed 2025-01-19 by 7wp81x