BruceDevices/firmware #672: LilyGo T-Display S3

OriginalNode, 2025-01-11:
Will you support this device ?

epiHATR, 2025-01-31:
now supported

bmorcelli, 2025-01-31:
We just need to find a way to make navigation with only two btns..

Short Right press> Next
Long right press >Select
Short left press>prev
Long left press>Esc

It would be nice.. and for the keyboard, think of a logic to use it too

(Can keep the 3rd btn logic, but if it can work with only 2, better..)

Grzybekrk, 2025-02-21:
Is working on amoled???

djaysan, 2025-02-23:
@bmorcelli that's awesome ! 
t-display s3 here, i understand that we need to short pin 16 & ground to select an option. 
How de we wire an sd card reader? 
 
I did that on t-display 1.14 following [this guide](https://github.com/ATOMNFT/Marauder-T-display-TTGO?tab=readme-ov-file#esp32-marauder-info) but i suppose Bruce is different and the pin numbers as well

bmorcelli, 2025-02-23:
I believe we need to make the SD, CC1101 and NRF pins configurable by software.. it would be easier to setup

Like having a field on bruce.conf, like 
'SD_pins: {SCK,Mosi,Miso,CS},
CC_pins: {SCK,Mosi,Miso,CS},
NRF_pins: {SCK,Mosi,Miso,CS}'

bmorcelli, 2025-02-24:
If it is using 8bit parallel, it shouldn't have the TFT_MOSI defined in the .ini or pinouts file..

Comment them out and it should work just fine

epiHATR, 2025-02-25:
Thank you so much, i'll try

IncursioHack, 2025-03-27:
@epiHATR  Could you please provide us with feedback if the situation has been corrected?

closed 2025-04-25 by OriginalNode