BruceDevices/firmware #675: cc1101 issues m5stickc plus2

Sawduzt567, 2025-01-11:
Whenever selecting the cc1101 in the module menu, my m5stick freezes until i unplug one of the wires, i know that my wiring is correct as the same wiring works for the NRF24 module. if i shutdown the device on the module selection page, it ends up choosing it sucessfully when i reboot it, but none of the features work, spectrum shows nothing apart from small "RF - Spectrum" text, and copying or playing back signals is broken. ive tried using another cc1101 module but i ended up with the same results, reflashing firmware, changing wires, all didnt solve the issue.

sk-oberon, 2025-01-19:
What firmware version are you using?

I tested with both latest firmware versions - Stable and Beta - and CC1101 module (wired as per [CC1101 on M5Stack Stick Cplus](https://raw.githubusercontent.com/pr3y/Bruce/refs/heads/main/media/connections/cc1101_stick.jpg)) works fine with my M5StickC PLUS2

killerbe, 2025-01-19:
Hey, all. 
Problem with SubGHZ. There is a call with a transmitter working on 433.92. The key is static, not dynamic. When saving, does not play the call, although when you just catch, you can play, but only when you save it does not work correctly, with barriers not sovsehm. 
Tried to fix the code, the only thing prescribed the key and the sequence of bits transmitted.

Unfortunately I can attach a file with captured data before updating the code. With the new more readable did not save. But even saving raw data, does not play, I do not understand what's wrong. Can you tell me please?

Example 1:
Filetype: Bruce SubGhz File
Version 1
Frequency: 433657024
Preset: 0
Protocol: RAW
RAW_Data: 5489 -18 298 -7 203 -7 202 -7 231 -7 214 -7 221 -6 5571 -7 418 -7 203 -7 202 -7 221 -7 205 -7 5703 -18 453 -7 205 -7 210 -7 212 -7 206 -7 5661 -7 256 -7 205 -7 203 -7 220 -7 205 -7 5869 -7 300 -7 205 -7 202 -7 221 -7 209 -7 5821 -18 393 -7 205 -7 203 -7 220 -6 207 -7 5720 -7 411 -7 205 -7 203 -7 221 -6 206 -7 5713 -18 397 -7 205 -7 202 -7 221 -7 208 -7 5714 -7 388 -7 204 -7 203 -7 221 -7 207 -7 5735 -18 351 -7 205 -7 203 -7 218 -7 212 -7 5758 -7 320 -7 205 -7 202 -7 221 -7 210 -7 5800 -18 234 -7 204 -7 203 -7 219 -7 205 -7 5882 -7 433 -7 205 -7 203 -13 212 -7 206 -7 5693 -18 430 -7 205 -7 211 -7 212 -7 205 -7 5684 -7 405 -7 205 -7 202 -7 219 -7 207 -7 5720 -18 276 -7 205 -7 203 -7 220 -7 206 -7 5837 -7 375 -6 205 -7 203 -7 220 -7 210 -7 5746 -18 261 -7 204 -7 203 -7 220 -7 206 -7 5853 -7 303 -7 205 -7 203 -7 220 -7 206 -7 5821 -18 343 -7 205 -7 203 -7 220 -7 206 -7 5770 -7 256 -7 205 -7 202 -7 221 -7 205 -7 5869 -18 298 -7 205 -7 203 -6 221 -7 206 -7 215 -6 226 -9 12 -13 12 -12 12 -12 13 -5290 6 -430 7 -205 7 -208 7 -212 7 -206 7 -214 7 -5462 18 -336 7 -203 7 -203 7 -220 7 -207 7 -215 6 -5557 7 -275 7 -203 7 -202 7 -5295 6 -354 7 -205 7 -202 7 -217 7 -207 7 -5769 7 -230 7 -205 7 -202 7 -215 7 -206 7 -5900 19 -373 7 -205 7 -202 7 -217 7 -208 7 -5741 7 -419 7 -204 7 -203 7 -216 7 -207 7 -5709 18 -403 7 -205 7 -202 7 -215 7 -207 7 -215 6 -226 9 -12 12 -13 12 -12 12 -12 5191 -7 389 -7 205 -7 203 -7 217 -7 205 -7 215 -7 5502 -7 373 -7 202 -7 203 -7 217 -7 209 -6 215 -7 5533 -19 272 -7 203 -7 202 -7 217 -7 208 -7 215 -6 5623 -7 367 -7 203 -7 202 -7 217 -7 209 -7 215 -6 5539 -19 361 -7 203 -7 202 -7 217 -7 208 -7 215 -7 5533 -8 431 -7 203 -7 202 -15 208 -7 204 -7 215 -7 5479 -18 251 -6 7 -221 7 -5548 7 -313 7 -203 7 -202 7 -223 7 -215 7 -221 6 -5575 18 -411 7 -203 7 -203 7 -222 7 -211 6 -222 6 -5470 7 -371 7 -202 7 -203 7 -231

Example 2:
Filetype: Bruce SubGhz File
Version 1
Frequency: 433919968
Preset: 1
Protocol: RcSwitch
Bit: 24
Key: я і	яя?і
TE: 249

Example 3:
Filetype: Bruce SubGhz File
Version 1
Frequency: 433919968
Preset: 1
Protocol: RcSwitch
Bit: 24
Key: я і	яя?і
TE: 249

sk-oberon, 2025-01-19:
@killerbe Are you using M5StickC PLUS2? If so, what firmware version are you using?

Symptoms you described fit the issue v1.8 has - Decoded RF signal using RcSwitch protocol was incorrectly saved in ASCII (Key: я і яя?і�) instead of a HEX value.

Either try beta version (contains fix) firmware or v1.8.1 (released 5 hours ago - no idea if it contains fix already).

killerbe, 2025-01-19:
Yes used M5StickC PLUS2, sold it, will it fit M5Stack Cores3 SE ?

> [@killerbe](https://github.com/killerbe) Are you using M5StickC PLUS2? If so, what firmware version are you using?
>
> Symptoms you described fit the issue v1.8 has - Decoded RF signal using RcSwitch [...]

bmorcelli, 2025-01-20:
> Yes used M5StickC PLUS2, sold it, will it fit M5Stack Cores3 SE ?
> 

Core devices still doesn't support CC1101

closed 2025-01-20 by bmorcelli

killerbe, 2025-02-05:
Module cc1101 does not work properly on frequency 433.92

Steps To Reproduce:
Caught a signal and saved it.
File - the signal did not play in the saved file.
It worked only when it was not saved.

Device used: M5StickPlus2
Filetype: Bruce SubGhz File
Version 1
Frequency: 433919968
Preset: 1
Protocol: RcSwitch
Bit: 24
Key: 00 00 00 00 00 D5 12 28
TE: 249

PanagossDev, 2025-02-28:
> Whenever selecting the cc1101 in the module menu, my m5stick freezes until i unplug one of the wires, i know that my wiring is correct as the same wiring works for the NRF24 module. if i shutdown [...]

I have the EXACT same problem brother! I am trying to find a solution and I still can't, I am about to lose my mind

aiqu-boop, 2025-04-23:
нашли решение?

Cboxolawd, 2025-06-24:
> I have the EXACT same problem brother! I am trying to find a solution and I still can't, I am about to lose my mind

Anyone found the solution by now?
I’m having this problem now.

Huzzla101, 2025-06-24:
If your using a m5stick cplus you should use the m5 module i think....
CC1101 module is ment for Lilygo etc that have internal cc1101 antenna....
When entering RF meny it should not say cc1101 module up top but "RADIO
FREQUENCY  Rxc pinout nr and Tx pinout nr  " then it will work .... If you
changes to cc1101 module it will distract your wire setup and get wrong
pins and then freeze or do a MAD lagg. Only make sure you put the correct
pins in config that you are using for Tx and Rx if they do not match ur
setup.  So as long as it says RADIO FREQ and not cc1101 on top when
entering RF meny, then dont go into modules at all but only change the pin
outs in config as if you enter module and then enter eather of them it may
shock antenna and give u lag again... Even if u enter m5 module  as it will
reset the pins to default and that may not fit your setup

BUT It will only work for default 433.92 mhz as CC1101 module is not made
yet for m5stick but as mentioned above.... So if you have a good 433RF
sender and reciever i would recomend that, it works VERY well and you can
enter the sender on right side no wires needed. and the reciever on the
Groove Hy2.0 left side same time  and aliexpress  ships the best quality
antennas for a dollar or 2 ... But CC1101 module will come for m5 also but
they have very little flash memory to use left etc..

Opolopopolo, 2025-06-29:
Thanks, the recommendation not to switch the setting to shared spi helped me get nrf and cc1101 up and running, but unfortunately I still can't get them to work with the sd card at the same time. I'm using stick2 with a Pirata expansion board.