BruceDevices/firmware #680: fail to mount sd card

Wesamalmsrwry, 2025-01-13:
**Describe the bug:**

sometimes sd card disappear or it says  fail to mount but sdcard plugged in
**Steps To Reproduce:**

randomly happens
- Device used: 
cardputer

**Additional context / logs:**

to solve this issue i have to reset device if it not solve it i have to remove the sdcard and insert it again

Wesamalmsrwry, 2025-01-14:
> > **Describe the bug:**
> > sometimes sd card disappear or it says fail to mount but sdcard plugged in **Steps To Reproduce:**
> > randomly happens
> > 
> > * Device used:
> >   cardputer
> > 
> > **Additional context / logs:**
> > to solve this issue i have to reset device if it not solve it i have to remove the sdcard and insert it again
> 
> Try sdcard that is small size bro.. mine doesn't works on 32GB sdcard... but when i tired it with 2GB it works well..

I`m using 512mb sdcard

7wp81x, 2025-01-18:
> I`m using 512mb sdcard

I guess it is a bug.. i tried also using 258MB sdcard and it works well with other firmware (ESP32 Marauder-CYD) also with the M5 Launcher.

wesamdev, 2025-01-22:
> I guess it is a bug.. i tried also using 258MB sdcard and it works well with other firmware (ESP32 Marauder-CYD) also with the M5 Launcher.

even with m5luancher somtimes not work i have to remove it and return it 
iam wesamalmasrwry on other account.

akeendev, 2025-01-31:
I'm having the same problem, but it only happens when I use the 'raw sniffer.' I'm using a 32GB SanDisk SD card

entroxx, 2025-02-18:
i have the same problem. I use a 64gb sandisk ultra sd card

7wp81x, 2025-02-19:
> i have the same problem. I use a 64gb sandisk ultra sd card

format it as fat32 then try creating 4GB partition or lower...

entroxx, 2025-02-19:
okay thanks