BruceDevices/firmware #681: Brucegotchi handshake bug

misterminefort, 2025-01-13:
**Describe the bug:**
Brucegotchi isnt fully capturing the handshakes, like i have message 1 and 2, but the rest didnt get captured.

**Steps To Reproduce:**
Start Brucegotchi
- Device used: M5StickC Plus2

**Expected behavior:**
Handshake having all the 4 messages

**Actual behavior:**
Handshake not having all the messages

**Additional context / logs:**
Heres a part of my log:
Selected: Brucegotchi
Brucegotchi Initialized
[222050][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[222056][W][sd_diskio.cpp:512] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[222066][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[222073][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[222583][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[222588][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
<<---- Starting Deauthentication Process ---->>
D8:32:14:07:B4:81 on ch1 -> we are now on ch 11
D8:32:14:07:B4:81 on ch11 -> we are now on ch 11
[226497][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[226513][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.

Preparing deauth frame to AP -> D8:32:14:7:B4:81 and Tgt: FF:FF:FF:FF:FF:FFD8:32:14:07:B4:82 on ch1 -> we are now on ch 11
D8:32:14:07:B4:82 on ch11 -> we are now on ch 11
[226921][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[226936][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.

Preparing deauth frame to AP -> D8:32:14:7:B4:82 and Tgt: FF:FF:FF:FF:FF:FF<<---- Stopping Deauthentication Process ---->>
<<---- Starting Deauthentication Process ---->>
D8:32:14:07:B4:81 on ch1 -> we are now on ch 11
D8:32:14:07:B4:81 on ch11 -> we are now on ch 11
[227492][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[227503][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.

Preparing deauth frame to AP -> D8:32:14:7:B4:81 and Tgt: FF:FF:FF:FF:FF:FFD8:32:14:07:B4:82 on ch1 -> we are now on ch 11
D8:32:14:07:B4:82 on ch11 -> we are now on ch 11
[227565][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[227582][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.

Preparing deauth frame to AP -> D8:32:14:7:B4:82 and Tgt: FF:FF:FF:FF:FF:FF<<---- Stopping Deauthentication Process ---->>
<<---- Starting Deauthentication Process ---->>
D8:32:14:07:B4:81 on ch1 -> we are now on ch 11
D8:32:14:07:B4:81 on ch11 -> we are now on ch 11
[228523][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[228534][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.

Preparing deauth frame to AP -> D8:32:14:7:B4:81 and Tgt: FF:FF:FF:FF:FF:FFD8:32:14:07:B4:82 on ch1 -> we are now on ch 11
D8:32:14:07:B4:82 on ch11 -> we are now on ch 11
[228596][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[228613][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.

Preparing deauth frame to AP -> D8:32:14:7:B4:82 and Tgt: FF:FF:FF:FF:FF:FF<<---- Stopping Deauthentication Process ---->>
<<---- Starting Deauthentication Process ---->>
D8:32:14:07:B4:81 on ch1 -> we are now on ch 11
D8:32:14:07:B4:81 on ch11 -> we are now on ch 11
[229503][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[229514][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.

Preparing deauth frame to AP -> D8:32:14:7:B4:81 and Tgt: FF:FF:FF:FF:FF:FFD8:32:14:07:B4:82 on ch1 -> we are now on ch 11
D8:32:14:07:B4:82 on ch11 -> we are now on ch 11
[229576][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[229593][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.

Preparing deauth frame to AP -> D8:32:14:7:B4:82 and Tgt: FF:FF:FF:FF:FF:FF<<---- Stopping Deauthentication Process ---->>
<<---- Starting Deauthentication Process ---->>
D8:32:14:07:B4:81 on ch1 -> we are now on ch 11
D8:32:14:07:B4:81 on ch11 -> we are now on ch 11
[230484][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[230495][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.

Ryvek, 2025-01-18:
damn

bmorcelli, 2025-01-18:
Capturing without an SD Card is really hard.. because writing on flash (LittleFs) is damn slower than on SD..

So this might be the problem.. doesn't have enough speed to process the packet/writing until the next come up

misterminefort, 2025-01-19:
> Capturing without an SD Card is really hard.. because writing on flash (LittleFs) is damn slower than on SD..
> So this might be the problem.. doesn't have enough speed to process the packet/writing until the next come up

Thanks @bmorcelli for pointing this out, I will go ahead and buy an SD card module so it can work.

closed 2025-01-19 by misterminefort