BruceDevices/firmware #690: cyd CYD-JC2432W328R

BblackmilkK, 2025-01-15:
**Describe the bug:**
<screen touch not worikng any version. cyd2usb totally not working.>

<i tried every steps>
- Device used: 
web flasher + cyd CYD-JC2432W328R

bmorcelli, 2025-01-15:
This device isn't supported yet.. only 328C for now

BblackmilkK, 2025-01-16:
thank you for answer. The screen is running(328c). But the touch does not work. Are you planning to do this?

BblackmilkK, 2025-01-16:
I would be very happy.

Ruslan228asr, 2025-01-16:
> I would be very happy.

Find projects where your display is supported, and you can throw this git here. And now there is a bug fix for 24R, and it will be hard for the developer to optimize the new version of CYD for your display.

BblackmilkK, 2025-01-16:
> > I would be very happy.
> 
> Find projects where your display is supported, and you can throw this git here. And now there is a bug fix for 24R, and it will be hard for the developer to optimize the new version of CYD for your display.

> This device isn't supported yet.. only 328C for now

thank you for answer. The screen is running(328c). But the touch does not work. Are you planning to do this? I would be very happy.

BblackmilkK, 2025-01-16:
> This device isn't supported yet.. only 328C for now

thank you for answer. The screen is running(328c). But the touch does not work. Are you planning to do this? I would be very happy.

bmorcelli, 2025-01-16:
Can you try the CYD-2USB build for us, please?

I couldn't find out which pins it uses for touchscreen, but they usually use the same XPT2046 driver, but I need to know in which pins is it connected 

If you come with this info, things can speed up

BblackmilkK, 2025-01-16:
> Can you try the CYD-2USB build for us, please?
> 
> I couldn't find out which pins it uses for touchscreen, but they usually use the same XPT2046 driver, but I need to know in which pins is it connected
> 
> If you come with this info, things can speed up

The screen doesn't even turn on when I use it. (cyd2usb) I could not find any information for 2432W328R pins ( only find : https://www.reddit.com/r/esp32/comments/1dy5k11/working_cyd_jc2432w328_display_240x320_28_usbc/  )

BblackmilkK, 2025-01-17:
> Can you try the CYD-2USB build for us, please?
> 
> I couldn't find out which pins it uses for touchscreen, but they usually use the same XPT2046 driver, but I need to know in which pins is it connected
> 
> If you come with this info, things can speed up

http://pan.jczn1688.com/directlink/1/HMI%20display/JC2432W328.zip?spm=a2g0s.imconversation.0.0.2a8b3e5fVptCV7&file=JC2432W328.zip

[image]

ZaabazingooO, 2025-01-17:
i need to. i have 2432W328R cheap yellow diplay.

bmorcelli, 2025-01-17:
Can you try installing the Beta from my repo??

I added an option for 2432W328R

Flasher is here https://bmorcelli.github.io/Bruce/flasher

ZaabazingooO, 2025-01-17:
> Can you try installing the Beta from my repo??
> 
> I added an option for 2432W328R
> 
> Flasher is here https://bmorcelli.github.io/Bruce/flasher

not working . i tried.

BblackmilkK, 2025-01-17:
> Can you try installing the Beta from my repo??
> 
> I added an option for 2432W328R
> 
> Flasher is here https://bmorcelli.github.io/Bruce/flasher

i trying i ll tell u

BblackmilkK, 2025-01-17:
Not working . does not open. first white screen hangs on the screen.

ZaabazingooO, 2025-01-17:
> Not working . does not open. first white screen hangs on the screen.

same for to me

bmorcelli, 2025-01-18:
New try on https://bmorcelli.github.io/Bruce/flasher

BblackmilkK, 2025-01-18:
> New try on https://bmorcelli.github.io/Bruce/flasher

is booting work. touch not work :(

ZaabazingooO, 2025-01-18:
> New try on https://bmorcelli.github.io/Bruce/flasher

screen tuouch not working

bernayilmaz921, 2025-01-18:
firmware booting work.
color is good.
tousch screen is not working.

ZaabazingooO, 2025-01-19:
I have this card and didnt work touch screen

kimseyokki, 2025-01-19:
I have this card and didnt work touch screen

ZaabazingooO, 2025-01-19:
> I have this card and didnt work touch screen

w r waiting to.

BblackmilkK, 2025-01-19:
developer will help us. we have to be patient.

bmorcelli, 2025-01-20:
Not yet..

I'm trying to make a lib work with no lucky.. It'll take some time

bmorcelli, 2025-01-20:
New try on https://bmorcelli.github.io/Bruce/flasher (BetaRelease(

I need Serial Logs when touching screen..

ZaabazingooO, 2025-01-20:
> New try on https://bmorcelli.github.io/Bruce/flasher (BetaRelease(
> 
> I need Serial Logs when touching screen..

I don't know how to do it.

BblackmilkK, 2025-01-20:
not working :( 

https://fr4nkfletcher.github.io/Adafruit_WebSerial_ESPTool/
this frimware touch screen works. but it doesn't touch the right places of the screen.

[image]

bmorcelli, 2025-01-20:
Nice... now I am very comfident that it will work:

 try on https://bmorcelli.github.io/Bruce/flasher (Beta Release)

BblackmilkK, 2025-01-20:
> Nice... now I am very comfident that it will work:
> 
> try on https://bmorcelli.github.io/Bruce/flasher (Beta Release)

no not working :(

kimseyokki, 2025-01-20:
Im looking forward to it

bmorcelli, 2025-01-21:
New Commit on https://bmorcelli.github.io/Bruce/flasher (Beta Release)

ZaabazingooO, 2025-01-21:
> Nice... now I am very comfident that it will work:
> 
> try on https://bmorcelli.github.io/Bruce/flasher (Beta Release)

i didint try this (previous) version. but i tryed last one and it is not booting

bernayilmaz921, 2025-01-22:
> > Nice... now I am very comfident that it will work:
> > try on https://bmorcelli.github.io/Bruce/flasher (Beta Release)
> 
> i didint try this (previous) version. but i tryed last one and it is not booting

m2 . I did not try previous but this version not booting

BblackmilkK, 2025-01-22:
> any news ?

we are waiting

ZaabazingooO, 2025-01-24:
1+

BblackmilkK, 2025-01-25:
:(

BblackmilkK, 2025-01-27:
[image]

BblackmilkK, 2025-01-27:
[Getting started JC2432W328.pdf](https://github.com/user-attachments/files/18552203/Getting.started.JC2432W328.pdf)

[image]

bmorcelli, 2025-01-27:
The last code I uploaded in my brench was supposed to work.. but I'm getting knocked out by the Watchdog timers...

I'm taking out the taskInputHandler and making it different (it brought many bugs instead of making things better).. once I get it working fine, I'll post here again

BblackmilkK, 2025-01-27:
> The last code I uploaded in my brench was supposed to work.. but I'm getting knocked out by the Watchdog timers...
> 
> I'm taking out the taskInputHandler and making it different (it brought many bugs instead of making things better).. once I get it working fine, I'll post here again

thank you.. i waiting.

ZaabazingooO, 2025-01-27:
> The last code I uploaded in my brench was supposed to work.. but I'm getting knocked out by the Watchdog timers...
> 
> I'm taking out the taskInputHandler and making it different (it brought many bugs instead of making things better).. once I get it working fine, I'll post here again

last version is was stack on boot loop

bmorcelli, 2025-01-27:
> last version is was stack on boot loop

Yes, I know..

ZaabazingooO, 2025-01-28:
> New try on https://bmorcelli.github.io/Bruce/flasher

boot is done..
touch is not :(

BblackmilkK, 2025-01-28:
> New try on https://bmorcelli.github.io/Bruce/flasher

touch not working :(

BblackmilkK, 2025-01-28:
> New try on https://bmorcelli.github.io/Bruce/flasher

The screen touch keys do not work, but when the screen turns off, it turns on when I touch it

ZaabazingooO, 2025-01-29:
> > New try on https://bmorcelli.github.io/Bruce/flasher
> 
> The screen touch keys do not work, but when the screen turns off, it turns on when I touch it

+1

BblackmilkK, 2025-01-31:
dont forget us :)

bmorcelli, 2025-01-31:
I need to have this board in hands in order to run some tests.. can't go further without it.. 

I couldn't find this board on AliExpress, and I need cash to buy it.. 

https://buymeacoffee.com/bmorcelliz

ZaabazingooO, 2025-02-01:
> I need to have this board in hands in order to run some tests.. can't go further without it..
> 
> I couldn't find this board on AliExpress, and I need cash to buy it..
> 
> https://buymeacoffee.com/bmorcelliz

https://www.aliexpress.com/item/1005006729951968.html?spm=a2g0o.order_detail.order_detail_item.5.4e33f19cayeGt9

Dtrieb, 2025-02-03:
Maybe this helps you:
https://github.com/rzeldent/esp32-smartdisplay

bmorcelli, 2025-02-15:
Hey,

Can you test here: https://buymeacoffee.com/bmorcelliz 
choose BetaRelease -> WIP -> CYD-2432W328R or 2432S024R"

BblackmilkK, 2025-02-16:
>Hey,
> 
> Can you test here: https://buymeacoffee.com/bmorcelliz 
> choose BetaRelease -> WIP -> CYD-2432W328R or 2432S024R"
> 
> 

 There is only a donation page.

bmorcelli, 2025-02-16:
Oops

https://bmorcelli.github.io/Bruce/flasher

bmorcelli, 2025-02-16:
we need to set up the calibration data for this device:

flash this binary, hit the corners shown by the arrows and send me the data that is printed at the screen (or serial monitor)

after the calibration, tap the screen at some pointsand make sure the red dot is drawn at the right places

Unzip and install it using https://web.esphome.io

[tft_calib.zip](https://github.com/user-attachments/files/18814677/tft_calib.zip)

Dtrieb, 2025-02-16:
> we need to set up the calibration data for this device:
>
> flash this binary, hit the corners shown by the arrows and send me the data that is printed at the screen (or serial monitor)
>
> after the [...]

```
[14:19:18]ets Jul 29 2019 12:21:46
[14:19:18]
[14:19:18]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:19:18]configsip: 0, SPIWP:0xee
[14:19:18]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:19:18]mode:DIO, clock div:2
[14:19:18]load:0x3fff0030,len:1184
[14:19:18]load:0x40078000,len:13232
[14:19:18]load:0x40080400,len:3028
[14:19:18]entry 0x400805e4
[14:19:19]E (165) esp_core_dump_flash: No core dump partit1�.2�չ��
[14:19:19]E (165) esp_core_dump_flash: No core dump partition found!
[14:19:19]E (169) esp_ota_ops: not found otadata
[14:19:19][     8][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[14:19:19][    35][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[14:19:19][    45][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[14:19:19][    56][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
```

Didn't show something on the screen.

BblackmilkK, 2025-02-16:
> ```
> [14:19:18]ets Jul 29 2019 12:21:46
> [14:19:18]
> [14:19:18]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> [14:19:18]configsip: 0, SPIWP:0xee [...]

+1

bmorcelli, 2025-02-17:
I added a Calibration process at the start of Bruxe, or this device, maybe it works now..

this calibration will be done one time, and to do it again you need to delete the "calData" file from LittleFs and restart..

can you test it, please?
https://bmorcelli.github.io/Bruce/flasher

BetaRelease -> WIP -> CYD-2432W328R or 2432S024R"

ZaabazingooO, 2025-02-17:
> I added a Calibration process at the start of Bruxe, or this device, maybe it works now..
>
> this calibration will be done one time, and to do it again you need to delete the "calData" file from [...]

Thank you for your effort ,working

closed 2025-02-17 by bmorcelli

bmorcelli, 2025-02-21:
Can you send me the content of the "callData" file, please?

BblackmilkK, 2025-02-21:
> Can you send me the content of the "callData" file, please?

me ?

[4 more replies]