BruceDevices/firmware #691: pl-PL keyboard layout

Endrive, 2025-01-15:
I wanted to point out that the Polish Bad BLE user community is really large, so adding the pl-PL keyboard layout would be a great solution.

bmorcelli, 2025-01-20:
Look into this page and tell us what is the most common arrangement: ANSI104, ISO105 or other?

https://kbdlayout.info/KBDPL

bmorcelli, 2025-01-20:
Note that special characters won't work... Only ASCII characters are supported

bmorcelli, 2025-01-20:
I added this key layout in my repo for testing 

New try on https://bmorcelli.github.io/Bruce/flasher (BetaRelease)

Endrive, 2025-01-23:
i will check

Tawank, 2025-01-29:
This layout does not work. The `Polish (214)` layout was used in old typewriters and early macs from the '90s.
Nowadays in Poland we use `Polish (Programmers)` layout (https://kbdlayout.info/00000415), which is `US-International` layout with additional special key combinations and I guess ANSI is more common.

I tested the `US-International` layout on a Polish keyboard, and it works fine. 
I guess we can delete the `pl-PL` layout to save memory, or maybe create an alias to `US-International`.

bmorcelli, 2025-01-29:
Available on Beta

closed 2025-01-29 by bmorcelli