BruceDevices/firmware #700: T-Embed CC1101 wrong battery level Bruce 1.8.1

lgstoian, 2025-01-19:
**Describe the bug:**
<In 1.8.1 the battery level for the T-Embed CC1101 is displayed incorrectly. Full charge shows up as 37%. This issue is not present if I flash back to 1.8.0.>

**Steps To Reproduce:**
<Booting the fully charged device shows 37% battery.>
- Device used: T-Embed CC1101

**Expected behavior:**
<Battery level should be 100%>

**Actual behavior:**
<Wrong battery level displayed.>

SteveHawk, 2025-01-21:
I'm also having the same issue. After fully charging the device, it only shows 52% battery. Leave the battery to fully drained then charge again, it still caps at 52%.

So it appears #560 works differently on individual units. Maybe it needs some sort of calibration?  @rouing could you help us look into the issue here?

ZaabazingooO, 2025-01-21:
CC1101 not working at all . on 1.8.1 m5stick cplus 2 and t-deck

Jiahao09, 2025-01-21:
Same here, when its at 0 it still able to use for another 10minutez?

realdaveblanch, 2025-01-22:
Brand new embed cc1101, batt charging shows 98% at maximum, not more than that (v 1.8.1)

cku-heise, 2025-01-23:
Brand new device here as well, after stock firmware and a short intermezzo with Capybara, flashed to Bruce. Fully charged battery (charged overnight) shows as 51%.

bmorcelli, 2025-01-29:
Fixed on Beta

closed 2025-01-29 by bmorcelli

realdaveblanch, 2025-02-19:
> Fixed on Beta

Proven to work perfectly