BruceDevices/firmware #702: BAD BLE causes screen glitch

Gosheto1234, 2025-01-20:
**Describe the bug:**
When I cancel the Bad BLE script after I run it the names in the main menu glitch and also the icons when I hold the next button 

**Steps To Reproduce:**

- Device used: m5stickplus2 

**Expected behavior:**
 I don't expect the text and icons to glitch 

**Actual behavior:**
yesterday when I downloaded the new firmware version on my m5stickplus2 , I added custom color UI , changed the BLE spam to be faster ( code changes) then added my other files through web ui and when I ran my BLE script I didn't had device that is connected and I canceled the script , when I got back to the main menu and I hold the next button on the stick , the icons and text were glitching , that was happening also when I am in a app , like wifi , ir etc. This is happening until I restart or power off and power on the m5stickplus2 

**Additional context / logs:**

bmorcelli, 2025-01-20:
Bluetooth is running on the background, so the drawings take a little more time to be drawn..

bmorcelli, 2025-01-20:
Now that I'm aware of it, probably..

closed 2025-05-24 by bmorcelli