BruceDevices/firmware #703: Lilygo T-Embed CC1101 can't write with keyboard interface

fkrlidon, 2025-01-20:
**Describe the bug:**
Lilygo T-Embed CC1101 can't write with keyboard interface

**Steps To Reproduce:**
For example: go to Files -> SD card -> bruce.conf -> rename. You can't type any character, only can use the buttons OK,CAP,DEL and SPACE.
- Device used: Lilygo T-Embed CC1101

**Expected behavior:**
After SPACE button, go to keyboard characters.

**Actual behavior:**
I can't select keyboard characters

**Additional context / logs:**
Version 1.8

bmorcelli, 2025-01-20:
In this version, hold the upper btn (the one you use to turn on) and roll the encoder wheel

In version 1.8.1 it behaves as you said.. and the esc logic works too

closed 2025-01-20 by bmorcelli

fkrlidon, 2025-01-20:
It's works. Thanks.