BruceDevices/firmware #705: Wifi deauth / deauth flood

naofumi39538, 2025-01-20:
When i updated the bruce in my m5 stickc plus 2

1.8 > 1.8.1

The wifi deauth is not working properly, it doesn't disconnecting any wifi, please fix this, this feature is more useful 😔 🙏

And also, i tried the media Cmds and connect it in iphone, i click the screenshot, it didnt work too

NikolayDachev, 2025-01-23:
Lwhen deeauth is running cardputer restart after ~ 1min .. I can say defenetly something is wrong with 1,8,1

bmorcelli, 2025-01-23:
Logs logs

NikolayDachev, 2025-01-24:
> Logs logs

how ? I cannot find anythin on cardpputer sd ? whne deauth start it take less than min and carputer restart .. Ican confirm same cardputer with 1.8.0 .. no isse

bmorcelli, 2025-01-24:
Serial Monitor logs

NikolayDachev, 2025-01-26:
Thank you, 

Here are some logs , I gain same behaivior ,deauth restrat cartputer ,
Let me know if I can help with something else 

```[ 72734][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Valu                                                                                                                                                                                               e = 255
Selected: Wifi Atks
Selected: Target Atks
[ 91901][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 91943][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Started
[ 91949][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[ 99380][V][WiFiGeneric.cpp:384] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 22
[ 99388][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 1 - SCAN_DONE
Selected: dako_n
Selected: Deauth
[105336][V][WiFiGeneric.cpp:344] _arduino_event_cb(): STA Stopped
S105348[]1[0D5]3[4W8i]F[iVG]e[nWeirFiicG.ecnpepr:i1c0.4c0p]p :_3e93v]e n_taCradlulibnaoc_ke(v)e:n tA_rcdbu(i)n:o  AEPv eSntta:r t3 e-d
 TA_STOP
[105361][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START
[105362][V][WiFiGeneric.cpp:396] _arduino_event_cb(): AP Stopped
0105374][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 11 - AP_STO[P1
 5374][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Started
[105387][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START

Preparing deauth frame to AP -> 48:A9:8A:39:35:AC and Tgt: FF:FF:FF:FF:FF:FF[115328][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 164
[120326][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 0
[133464][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
[143472][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 164
[148470][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 0
[154768][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
[164776][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 164
[169774][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 0
[176311][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
[186319][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 164
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception).
Debug exception reason: Stack canary watchpoint triggered (InputHandler)
Core  0 register dump:
PC      : 0x40385faf  PS      : 0x00060436  A0      : 0x803842c8  A1      : 0x3fcede70
A2      : 0x3fc9f570  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060423
A6      : 0x00060423  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x3fcef920
A14     : 0x02c9f570  A15     : 0x00ffffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x40385fac:0x3fcede70 0x403842c5:0x3fcedeb0 0x40382604:0x3fcedee0 0x403825fa:0xa5a5a5a5 |<-CORRUPTED

ELF file SHA256: f61bc80a334449c4

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378b94
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x518
load:0x403c9700,len:0xc08
load:0x403cc700,len:0x2a0c
entry 0x403c98dc
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
[   612][D][esp32-hal-tinyusb.c:680] tinyusb_enable_interface(): Interface HID enabled
[   619][D][USBHID.cpp:61] tinyusb_enable_hid_device(): Device[0] len: 79
[   626][D][USBHID.cpp:61] tinyusb_enable_hid_device(): Device[1] len: 67
E (658) esp_ota_ops: not found otadata
[   691][D][main.cpp:298] setup(): Total heap: 248056
[   696][D][main.cpp:299] setup(): Free heap: 208316
[   700][D][main.cpp:302] setup(): PSRAM Not Found
[   705][D][main.cpp:303] setup(): Total PSRAM: 0
[   709][D][main.cpp:304] setup(): Free PSRAM: 0
[   714][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!
[... 93 lines cut ...]
[  9737][E][config.cpp:115] fromFile(): Fail
[  9742][E][config.cpp:121] fromFile(): Fail
[  9746][E][config.cpp:127] fromFile(): Fail
[  9750][E][config.cpp:129] fromFile(): Fail
[  9754][E][config.cpp:130] fromFile(): Fail
[  9758][E][config.cpp:132] fromFile(): Fail
[  9762][E][config.cpp:133] fromFile(): Fail
[  9766][E][config.cpp:134] fromFile(): Fail
[  9770][E][config.cpp:135] fromFile(): Fail
[  9774][E][config.cpp:136] fromFile(): Fail
[  9778][E][config.cpp:137] fromFile(): Fail
[  9782][E][config.cpp:139] fromFile(): Fail
[  9786][E][config.cpp:144] fromFile(): Fail
[  9790][E][config.cpp:146] fromFile(): Fail
[  9794][E][config.cpp:148] fromFile(): Fail
[  9798][E][config.cpp:149] fromFile(): Fail
[  9802][E][config.cpp:150] fromFile(): Fail
[  9806][E][config.cpp:158] fromFile(): Fail
[  9810][E][config.cpp:170] fromFile(): Fail to load qrCodes
[  9816][E][vfs_api.cpp:24] open(): File system is not mounted
[  9822][E][config.cpp:187] saveFile(): Failed to open config file
[  9828][I][config.cpp:176] fromFile(): Using config from file
[ 19662][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 164
[ 24670][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 0
```

NikolayDachev, 2025-01-26:
Also, I erase flash and reinstall  M5Launcher Cardputer also bruce 1.8.1 
same resultes, here are some more detialed logs , including  booting , deauth issue restart 

Also  it is posible deauth to be paused as well when computer screen trun off (power save ) and restrore when screen back on ... ?

I'm refering to those 

```
[ 69689][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 0
``
 
but may be is something normal so just ignore it .. however .. deauth restrat still persist 

```
[  1546][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - UsingPress the button                                                                                                                                                                                                to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x420f2f62
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x518
load:0x403c9700,len:0xc08
load:0x403cc700,len:0x2a0c
entry 0x403c98dc
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
[   614][D][esp32-hal-tinyusb.c:680] tinyusb_enable_interface(): Interface HID enabled
[   622][D][USBHID.cpp:61] tinyusb_enable_hid_device(): Device[0] len: 79
[   629][D][USBHID.cpp:61] tinyusb_enable_hid_device(): Device[1] len: 67
E (638) esp_ota_ops: not found otadata
[   651][D][main.cpp:298] setup(): Total heap: 248104
[   656][D][main.cpp:299] setup(): Free heap: 208896
[   660][D][main.cpp:302] setup(): PSRAM Not Found
[   665][D][main.cpp:303] setup(): Total PSRAM: 0
[   669][D][main.cpp:304] setup(): Free PSRAM: 0
[   674][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!
[  1919][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x42128824 arg=0x3fc9d8dc
[  1974][I][config.cpp:176] fromFile(): Using config from file
[  1998][I][config.cpp:176] fromFile(): Using config from file
[  2030][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
{
  "priColor": "a80f",
  "secColor": "880f",
  "bgColor": "0",
  "rot": 1,
  "dimmerSet": 10,
  "bright": 100,
  "tmz": 0,
  "soundEnabled": 1,
  "wifiAtStartup": 0,
  "ledBright": 75,
  "ledColor": "0",
  "webUI": {
    "user": "admin",
    "pwd": "bruce"
  },
  "wifiAp": {
    "ssid": "BruceNet",
    "pwd": "brucenet"
  },
[... 151 lines cut ...]
      "content": "WIFI:T:WPA;S:BruceNet;P:brucenet;;"
    },
    {
      "menuName": "Bruce Wiki",
      "content": "https://github.com/pr3y/Bruce/wiki"
    },
    {
      "menuName": "Bruce Site",
      "content": "https://bruce.computer"
    },
    {
      "menuName": "Rickroll",
      "content": "https://youtu.be/dQw4w9WgXcQ"
    }
  ]
}[  2110][I][config.cpp:195] saveFile(): config file written successfully
[  2142][W][LittleFS.cpp:77] begin(): LittleFS Already Mounted!
[  2205][E][vfs_api.cpp:105] open(): /littlefs/boot.jpg does not exist, no permits for creation
[  2222][E][vfs_api.cpp:105] open(): /littlefs/boot.gif does not exist, no permits for creation
[  9260][E][vfs_api.cpp:105] open(): /littlefs/boot.wav does not exist, no permits for creation
[  9514][I][config.cpp:176] fromFile(): Using config from file
[ 19468][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 164
[ 24476][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 0

```

NikolayDachev, 2025-01-26:
Intersting wtih c +2 no issues  , so can be something specific to cardputer, can be something specific when m5louncher is installed (not very posible) or something very specify with my cardputer (hmm :|) but in other point of view 0 issues with 1.8.0 "~

regards, let me know if I can help or more info is needed or sugersion for othe tyope  testing

bmorcelli, 2025-01-28:
> Debug exception reason: Stack canary watchpoint triggered (InputHandler)

Stack is too small for Cardputer in the loopInputTask, that inits in the main()

bmorcelli, 2025-01-28:
Can you test the BetaRelease from my repo?
 https://bmorcelli.github.io/Bruce/flasher

NikolayDachev, 2025-02-04:
> Can you test the BetaRelease from my repo? https://bmorcelli.github.io/Bruce/flasher

@bmorcelli  thank you!
Is there any way to download it ? I'm  using  launcher

NikolayDachev, 2025-02-04:
@bmorcelli thank you !

closed 2025-02-04 by bmorcelli