BruceDevices/firmware #709: m5stickc plus 2  not working cc1101 1.8.1

ZaabazingooO, 2025-01-21:
cc1101 not working on 1.8.1 but working before with 1.8

m5stickc plus 2

kimseyokki, 2025-01-21:
+ CYD ....028 it is not take any signal.

sk-oberon, 2025-01-21:
@ZaabazingooO What exactly doesn't work for you?
(The more details issue description contains the better)

Did you choose "CC1101 on SPI" in RF -> Config -> RF module?

CC1101 module works just fine with my M5StickC PLUS2 with FW v1.8.1.

[image]

ZaabazingooO, 2025-01-22:
> [@ZaabazingooO](https://github.com/ZaabazingooO) What exactly doesn't work for you? (The more details issue description contains the better)
> 
> Did you choose "CC1101 on SPI" in RF -> Config -> RF module?
> 
> CC1101 module works just fine with my M5StickC PLUS2 with FW v1.8.1.
> 
> [image]

I tried with cathack firmware and it worked. It doesn't work with this firmware.

killerbe, 2025-01-22:
> cc1101 not working on 1.8.1 but working before with 1.8
> 
> m5stickc plus 2

Hi. Did you solve the Bruce firmware problem after all?
Which module you are using can you attach a photo please.
And if you solved the problem, who helped you?

ZaabazingooO, 2025-01-22:
> > cc1101 not working on 1.8.1 but working before with 1.8
> > m5stickc plus 2
> 
> Hi. Did you solve the Bruce firmware problem after all? Which module you are using can you attach a photo please. And if you solved the problem, who helped you?

i used cathack cc1101 is working . when i change firmware to bruce cc1101 not working. but nrf24 is workig

Valentinlzr, 2025-01-23:
Tout pareil pour moi bruce 1.8.1 ne fonctionne pas sur m5stickc plus2 je reste bloquer sur booting

ZaabazingooO, 2025-01-25:
> Guys, which CC1101 module version are you using - v1 (10 pin) or v2 (8 pin) ?

8

Valentinlzr, 2025-01-25:
Yes 8 broche sur le m5stick c plus2. Enfait 1.8.1 fonctionne mais en
téléchargement direct via bruce ou sur m5luncher mais impossible via
m5launcher le fichier bin ne s'installe pas correctement. Mais la dernière
version de m5launcher est un peu instable sur le m5stick c plus2

closed 2025-01-29 by bmorcelli