BruceDevices/firmware #714: T go embed cc1101 buggy RF SPECTRUM

realdaveblanch, 2025-01-22:
**Describe the bug:**

Brand new (ver 1.8.1) after boot, click RF, SPECTRUM, after 1sec it reboots the device
**Steps To Reproduce:**

- Device used: Lily go t embed cc1101

- Click RF
-Click Scpectrum

**Expected behavior:**

- Show actual spectrum

**Actual behavior:**

- Reboots the device (can't show logs right now) (will provide if needed)

**Additional context / logs:**

bmorcelli, 2025-01-22:
Logs are important in this case

realdaveblanch, 2025-01-22:
Tried... Won't show as serial interface on Debian, only shows as hid device when using clicker feature, will try tomorrow on Windows

realdaveblanch, 2025-01-22:
> Logs are important in this case

Selected: Read tag
[ 25938][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[ 25945][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[ 35937][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 21 - Using Channel 7, Value = 13
[ 39535][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 21 - Using Channel 7, Value = 255
Selected: Spectrum
cc1101 Connection OK
cc1101 setMHZ(frequency);
cc1101 SetRx();
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x40382a74
file: "src/modules/rf/rf.cpp" line 72
func: void initRMT()
expression: rmt_driver_install(rxconfig.channel, 2048, 0)

abort() was called at PC 0x40382a77 on core 1

Backtrace: 0x403790ae:0x3fceb7c0 0x40382a81:0x3fceb7e0 0x40389389:0x3fceb800 0x40382a77:0x3fceb880 0x42054315:0x3fceb8a0 0x42056c62:0x3fceb8f0 0x4201565f:0x3fceb930 0x4200e585:0x3fceb950 0x42015af6:0x3fceb9c0 0x4225f4b9:0x3fcebc90 0x42031b11:0x3fcebcb0 0x421285ec:0x3fcebd60

ELF file SHA256: 91d746726a624cc2

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42128fa7
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   613][D][esp32-hal-tinyusb.c:680] tinyusb_enable_interface(): Interface HID enabled
[   621][D][USBHID.cpp:61] tinyusb_enable_hid_device(): Device[0] len: 79
[   627][D][USBHID.cpp:61] tinyusb_enable_hid_device(): Device[1] len: 67
E (651) esp_ota_ops: not found otadata
[   652][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   667][D][main.cpp:298] setup(): Total heap: 245504
[   672][D][main.cpp:299] setup(): Free heap: 211268
[   677][D][main.cpp:300] setup(): PSRAM Started
[   681][D][main.cpp:301] setup(): PSRAM Found
[   685][D][main.cpp:303] setup(): Total PSRAM: 8386215
[   690][D][main.cpp:304] setup(): Free PSRAM: 8381855
[   695][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=8 scl=18 freq=100000
[   704][I][XPowersCommon.tpp:445] begin(): SDA:8 SCL:18
[   709][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
getInputCurrentLimit: 3250 mA
getChargerConstantCurr: 832 mA
[  1302][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1408][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1514][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1620][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  1626][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  1632][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1641][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1747][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1853][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1959][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  1966][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2072][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2178][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2284][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  2290][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  2296][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  2305][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2411][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2517][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2623][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  2653][I][config.cpp:176] fromFile(): Using config from file
[... 375 lines cut ...]
[  3821][W][sd_diskio.cpp:174] sdCommand(): no token received
[  3927][W][sd_diskio.cpp:174] sdCommand(): no token received
[  4033][W][sd_diskio.cpp:174] sdCommand(): no token received
[  4139][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  4150][E][vfs_api.cpp:113] exists(): File system is not mounted
[  4158][E][vfs_api.cpp:105] open(): /littlefs/boot.jpg does not exist, no permits for creation
[  4167][E][vfs_api.cpp:113] exists(): File system is not mounted
[  4175][E][vfs_api.cpp:105] open(): /littlefs/boot.gif does not exist, no permits for creation
[ 10637][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 21 - Using Channel 7, Value = 13
[ 11218][E][vfs_api.cpp:113] exists(): File system is not mounted
[ 11226][E][vfs_api.cpp:105] open(): /littlefs/boot.wav does not exist, no permits for creation
[ 11435][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 21 - Using Channel 7, Value = 255
[ 11666][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 11772][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 11878][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 11984][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 11990][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[ 11996][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 12005][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 12111][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 12217][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 12323][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 12369][I][config.cpp:176] fromFile(): Using config from file
[ 21435][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 21 - Using Channel 7, Value = 13
[ 26443][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 21 - Using Channel 7, Value = 0

varenichek22, 2025-01-27:
same issue

bmorcelli, 2025-01-28:
> rmt_driver_install(rxconfig.channel, 2048, 0)

This failure is happening due to the LED init, that inits all channels of RMT.. so when spectrum tries to make use of this driver, it is occupied.. I'm working on a fix for that

varenichek22, 2025-01-28:
Spent all day trying to turn off these leds when spectrum is on, but nothing things to work, or mb i am just noob and should study more.
Turning off Leds permanently helped. crutch :)

bmorcelli, 2025-01-28:
Yes.. you can comment this line: https://github.com/pr3y/Bruce/blob/7a7744700f059c933398265508d825b8248ec806/src/main.cpp#L329

I'm trying a few things here, with quite a success, need to test it a little more

bmorcelli, 2025-01-28:
Can you test the BetaRelease from my repo?
 https://bmorcelli.github.io/Bruce/flasher

realdaveblanch, 2025-01-28:
> Can you test the BetaRelease from my repo? https://bmorcelli.github.io/Bruce/flasher

WORKS!But crashes when i try to go back: 

Selected: Spectrum
cc1101 Connection OK
cc1101 setMHZ(frequency);
cc1101 SetRx();
cc1101 Connection OK
cc1101 setMHZ(frequency);
cc1101 SetRx();
E (31748) rmt: RMT RX BUFFER FULL
E (31945) rmt: ---RX buffer too small: 4
E (31946) rmt: ---RX buffer too small: 4
E (31957) rmt: ---RX buffer too small: 4
E (31970) rmt: ---RX buffer too small: 4
E (32001) rmt: ---RX buffer too small: 4
E (32032) rmt: ---RX buffer too small: 4
E (32053) rmt: RMT RX BUFFER FULL
E (32263) rmt: RMT RX BUFFER FULL
E (32436) rmt: RMT RX BUFFER FULL
E (32572) rmt: RMT RX BUFFER FULL
E (32826) rmt: RMT RX BUFFER FULL
E (32961) rmt: ---RX buffer too small: 4
E (32961) rmt: ---RX buffer too small: 4
E (32962) rmt: ---RX buffer too small: 4
E (32968) rmt: RMT RX BUFFER FULL
E (33067) rmt: RMT RX BUFFER FULL
E (33190) rmt: ---RX buffer too small: 4
E (33219) rmt: ---RX buffer too small: 4
E (33253) rmt: ---RX buffer too small: 4
E (33280) rmt: RMT RX BUFFER FULL
E (33455) rmt: RMT RX BUFFER FULL
E (33591) rmt: RMT RX BUFFER FULL
E (33722) rmt: RMT RX BUFFER FULL
E (33827) rmt: ---RX buffer too small: 4
E (33828) rmt: ---RX buffer too small: 4
E (33828) rmt: ---RX buffer too small: 4
E (33829) rmt: ---RX buffer too small: 4
E (33844) rmt: ---RX buffer too small: 4
E (33852) rmt: RMT RX BUFFER FULL
E (34010) rmt: ---RX buffer too small: 4
E (34010) rmt: ---RX buffer too small: 4
E (34011) rmt: ---RX buffer too small: 4
E (34012) rmt: ---RX buffer too small: 4
E (34013) rmt: ---RX buffer too small: 4
E (34017) rmt: ---RX buffer too small: 4
E (34021) rmt: ---RX buffer too small: 4
E (34031) rmt: ---RX buffer too small: 4
E (34050) rmt: RMT RX BUFFER FULL
E (34178) rmt: RMT RX BUFFER FULL
E (34380) rmt: RMT RX BUFFER FULL
E (34516) rmt: RMT RX BUFFER FULL
E (34676) rmt: RMT RX BUFFER FULL
E (34783) rmt: RMT RX BUFFER FULL
E (34852) rmt: RMT RX BUFFER FULL
E (34980) rmt: RMT RX BUFFER FULL
E (35101) rmt: RMT RX BUFFER FULL
E (35231) rmt: RMT RX BUFFER FULL
E (35322) rmt: RMT RX BUFFER FULL
E (35422) rmt: ---RX buffer too small: 4
E (35433) rmt: ---RX buffer too small: 4
E (35434) rmt: ---RX buffer too small: 4
E (35435) rmt: ---RX buffer too small: 4
E (35435) rmt: ---RX buffer too small: 4
E (35456) rmt: RMT RX BUFFER FULL
E (35691) rmt: RMT RX BUFFER FULL
E (35829) rmt: RMT RX BUFFER FULL
E (35960) rmt: RMT RX BUFFER FULL
E (36100) rmt: RMT RX BUFFER FULL
CORRUPT HEAP: Bad tail at 0x3fcc4790. Expected 0xbaad5678 got 0x000d8008

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x403790aa:0x3fceb920 0x40382a95:0x3fceb940 0x4038949d:0x3fceb960 0x403890b5:0x3fceba90 0x40379901:0x3fcebab0 0x403894cd:0x3fcebad0 0x4212dd1d:0x3fcebaf0 0x420536e6:0x3fcebb30 0x4205624f:0x3fcebb50 0x42015743:0x3fcebb90 0x4200e621:0x3fcebbb0 0x42015bda:0x3fcebc20 0x4225eee1:0x3fcebef0 0x42031ba5:0x3fcebf10 0x42127df8:0x3fcebfc0

ELF file SHA256: fb6275d506feec28

Rebooting...

varenichek22, 2025-01-29:
Works well. Sometimes same thing as @realdaveblanch happens, but it reloads and continues to work. 
I didn't find a difference between the USB cable connection and the disconnected, it works just as well.
Also, the square spectrum is strange, still don`t understand it`s purpose, will made my own :/
Thank you!

realdaveblanch, 2025-01-29:
> Works well. Sometimes same thing as [@realdaveblanch](https://github.com/realdaveblanch) happens, but it reloads and continues to work. I didn't find a difference between the USB cable connection [...]

Overall the actual problemi is solved, but it needs more refinement. 

UNNEDED INFO:

I just found out that it uses the same hardware as Flipper besides the CPU, i think we can collab to make a better software than Flipper

bmorcelli, 2025-01-29:
> I just found out that it uses the same hardware as Flipper besides the CPU, i think we can collab to make a better software than Flipper

This is the Idea.. lol... Flipper has an antenna circuit that can handle all ranges, like T-Embed CC1101, the other blue modules we usually use are designed to 433Mhz only...  I wish I could find the T-Embed CC1101 chip to bu.

The good part is that Bruce is evolving at a good pace

closed 2025-01-29 by bmorcelli

realdaveblanch, 2025-02-19:
> This is the Idea.. lol... Flipper has an antenna circuit that can handle all ranges, like T-Embed CC1101, the other blue modules we usually use are designed to 433Mhz only...  I wish I could find [...]

Actually does! 
It's the highest evolving fw of all