BruceDevices/firmware #715: Add bulk RFID keys

realdaveblanch, 2025-01-22:
**Is your feature request related to a problem? Please describe.**
Np

**Describe the solution you'd like**

Might be possible to add bulk rf keys (Mifare..) ? 

**Describe alternatives you've considered**

Build from source adding Mifare keys from some public git

**Additional context**

 ;) you guys are doing an absolute great job with this, might try to add my own methods and share

rennancockles, 2025-01-22:
It was added in the latest version. Bruce now contains some more keys other than default one. You can also add other keys to the bruce.conf file.

closed 2025-01-22 by rennancockles

8ight9ine3three, 2025-03-04:
Hi, I am trying to add more than 1 custom key to the config file. I have edited the config file to include a long list of custom MIFARE keys, yet when I restart Bruce w/ the new config file, all the custom keys disappear.

I am adding each key inbetween closed quotations as follows:

 "mifareKeys": ["FFFFFFFFFFFF"
  "A0A1A2A3A4A5"
  "D3F7D3F7D3F7"
  "000000000000"
  "B0B1B2B3B4B5"],

and I also tried without quotations

 "mifareKeys": [FFFFFFFFFFFF
  A0A1A2A3A4A5
],