BruceDevices/firmware #721: Add timezone

nikita17-py, 2025-01-23:
Please add timezone Almaty Kazakhstan, I will be very glad

rouing, 2025-01-23:
Related #543 #616, NTP/RTC/Time Refactor is planned and coming after HAL lib feature is staged.

nikita17-py, 2025-01-27:
> Related [#543](https://github.com/pr3y/Bruce/issues/543) [#616](https://github.com/pr3y/Bruce/issues/616), NTP/RTC/Time Refactor is planned and coming after HAL lib feature is staged.

at Cardputer I am not RTC module

rouing, 2025-01-27:
> > Related [#543](https://github.com/pr3y/Bruce/issues/543) [#616](https://github.com/pr3y/Bruce/issues/616), NTP/RTC/Time Refactor is planned and coming after HAL lib feature is staged.
> 
> at Cardputer I am not RTC module

I also mentioned NTP and Time. Either way this affects all time keeper functions.

closed 2025-05-26 by bmorcelli