BruceDevices/firmware #724: RF Spectrum Bug T-Embed CC1101

tadiyusss, 2025-01-24:
Device: T-Embed CC1101 
Firmware Version: c4cf5c7

When I try to use RF Spectrum it restarts and give this output in the logs. 

```
Selected: Spectrum
cc1101 Connection OK
cc1101 setMHZ(frequency);
cc1101 SetRx();
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x40382a74
file: "src/modules/rf/rf.cpp" line 72
func: void initRMT()
expression: rmt_driver_install(rxconfig.channel, 2048, 0)

abort() was called at PC 0x40382a77 on core 1

Backtrace: 0x403790ae:0x3fcebb40 0x40382a81:0x3fcebb60 0x40389389:0x3fcebb80 0x40382a77:0x3fcebc00 0x42054319:0x3fcebc20 0x42056c66:0x3fcebc70 0x4201565f:0x3fcebcb0 0x4200e585:0x3fcebcd0 0x42015af6:0x3fcebd40 0x4225f4bd:0x3fcec010 0x42031b15:0x3fcec030 0x421285f0:0x3fcec0e0

ELF file SHA256: cdbca6f5de4b358b

Rebooting...
```

bmorcelli, 2025-01-27:
This issue is related to the encoder LEDs feature.. removing this feature, RMT is able to start without bugs

varenichek22, 2025-01-27:
> This issue is related to the encoder LEDs feature.. removing this feature, RMT is able to start without bugs

how to fix this issue? (i am a beginner, sorry)

bmorcelli, 2025-01-29:
Solved in Beta

closed 2025-01-29 by bmorcelli