BruceDevices/firmware #726: Lilygo t-display

Grzybekrk, 2025-01-24:
Hi I have 2 boards lilygo t-display one is T-display s3Amoled 2.0 bruce is not working and Displays
LILYGO T-Display-S3 1.9-inch LCD display this one works good and im wondering if you have some list supported s3 displays from lilygo. 
Next questions are you planning add functions for CC1101 or rfid for them if yes when and where I can fond info about pins ??

Grzybekrk, 2025-02-13:
[image]

[image]

Please advise ??

Grzybekrk, 2025-02-21:
Pls

djaysan, 2025-02-23:
the flasher will list all the supported device here: https://bruce.computer/flasher